Ultimaker/Cura #2875: trimesh

fieldOfView, 2017-11-29:
When investigating the possibilities of creating manual supports and doing booleans in the front-end, I came across trimesh:
http://trimesh.readthedocs.io/en/latest/
Trimesh offers a lot of interesting functionality, such as splitting compound meshes into multiple watertight meshes, checking watertightness, determining the center of gravity (layflat++) vertex picking (painting), in addition to things that UM already does (with varying degrees of success) such as handling a transformation tree, calculating convex hulls, etc. Really interesting stuff, and it seems to be well maintained:
https://github.com/mikedh/trimesh

Like UM.Math and UM.Scene it is based on numpy and scipy, and there is obviously a lot of overlap between trimesh and the part of Uranium that deals with meshes. At first I was considering asking if there's any chance of TriMesh being included with Cura installs, so plugins could start using it. But looking into the overlap, it could also be interesting to consider porting UM to use trimesh in the background (with the current Uranium API intact).

Has trimesh ever been on the radar?

trimesh has three hard dependecies (numpy, scipy, networkx), two of which are already Uranium dependecies (numpy, scipy).

ChrisTerBeke, 2017-11-29:
Functionality seems nice and the repo looks very well maintained and documented indeed :)

nallath, 2017-11-30:
Trimesh isn't something we've ever looked at.

fieldOfView, 2018-08-17:
Now that Trimesh is in, shall I close this issue?

Or, we can keep it open to discuss the cool things we will be doing with it. I am working on a plugin that checks for watertightness and can split compound meshes. I am running into an issue though converting a trimesh back to valid UM MeshData for Cura to use, Has anyone done that succesfully?

The theory is easy:
```
    def _toTriMesh(self, mesh_data: MeshData) -> trimesh.base.Trimesh:
        return trimesh.base.Trimesh(vertices=mesh_data.getVertices(), faces=mesh_data.getIndices(), vertex_normals=mesh_data.getNormals())

    def _toMeshData(self, trinode: trimesh.base.Trimesh) -> MeshData:
        return MeshData(vertices=trinode.vertices, indices=trinode.faces, normals=trinode.vertex_normals)
```
However the resulting mesh is very broken in Cura (to the point that it hardly renders anything at all).

ChrisTerBeke, 2018-08-17:
There's an internal ticket as well to use trimesh for more things in the future, so this ticket might be closed indeed.

closed 2018-08-17 by fieldOfView