Ultimaker/Cura #2876: Speed settings ignored extruder one

woofoo33, 2017-11-29:
Cura 3.0.4 
Ultimaker 3 on the latest UK version,

The issue or setting problem is I set the speed to say 50mms after the 1st layer the head moves so fast it under extrudes, Yes I know the 1st layer is always slower as I see this in the settings, 

put the exact same settings into extruder 2 it work correct every time ....

I have had this problem for a while its as if EX 1 has a built in default speed after the 1st layer

Ghostkeeper, 2017-12-13:
I can't reproduce this bug. If I reduce the speed on the first extruder it prints slower (or at least the `F` parameter of the moves is lower).

Perhaps you could upload a project file where you changed the speed settings and expect it to appear in the g-code? Those project files contain all your settings so we can more easily reproduce it.

woofoo33, 2017-12-13:
just to add the the filament used was 
Nylon or flex I was having the issues with but on in Extuder 1 , 

Ill try to re create

ianpaschal, 2018-03-26:
Needs more info; never got it; closing.

closed 2018-03-26 by ianpaschal