Ultimaker/Cura #2877: Adaptive layers - top layer not shown in display?

smartavionics, 2017-11-29:
Just playing with this interesting new option and I notice that using the default adaptive layer params, the top layer(s) are not displayed when adaptive layers is enabled:
[image]
Not enabled:
[image]

smartavionics, 2017-11-29:
The top skin layers are there so it looks like that the model has been truncated rather than just a display problem.

smartavionics, 2017-11-29:
Here's the model, it's large so I am scaling it to 10%.
[new_moai.zip](https://github.com/Ultimaker/Cura/files/1513587/new_moai.zip)

ChrisTerBeke, 2017-11-29:
Yeah there's some weird issue that sometimes it misses a few layers. Haven't been able to pin it down yet. Maybe you can have a quick look at the code changes to see if it's something we missed?

smartavionics, 2017-11-29:
I'm browsing the code to see if I can find the problem. One thing I did notice is that on line 137 of FffPolygonGenerator.cpp you are doing `adaptive_layer_heights->getLayers()` even if adaptive_layer_heights is null.

smartavionics, 2017-11-29:
I have fixed the top of the head being missing but have not fixed the potential null pointer reference that I mention above. I leave that to you to do.

smartavionics, 2017-11-29:
[image]

smartavionics, 2017-11-30:
Just printing the above moai with spiralization and adaptive layers. There's no way this would spiralize well without using thin layers, just look at that chin!!

[image]

smartavionics, 2017-11-30:
Well he almost came out perfect except that something terrible has happened to the start/end gcode inclusion which caused the end gcode not to be executed and so the hot end just sat there and burnt a hole in the top of his head!  

How did this happen:

````
;TIME_ELAPSED:1591.253817
G1 F1800 E-6
M82 ; absolute extrusion mode
M107
b'M83 ; extruder relative mode\nG1 E-5 F2000 ; retract\nM104 S0 ; turn off extruder\nM140 S0 ; turn off bed\nM106 S0 ; turn off fan\nG28 ; home\nM84 ; disable motors\n'
M83 ; relative extrusion mode
M104 S0
;End of Gcode

````

Why is the end (and start) gcode prefixed with b and surrounded in quotes????

nallath, 2017-11-30:
Probably because somehwere bytes were converted to a string, which went wrong.

smartavionics, 2017-11-30:
Yes, this is what is being passed to CuraEngine:

````
-s machine_end_gcode="b'M83 ; extruder relative mode\nG1 E-5 F2000 ; retract\nM104 S0 ; turn off extruder\nM140 S0 ; turn off bed\nM106 S0 ; turn off fan\nG28 ; home\nM84 ; disable motors\n'"
````

smartavionics, 2017-11-30:
Hope someone fixes this as it rather stops me printing!

fieldOfView, 2017-11-30:
Looks like it is caused by this commit: 1d6b7c71cbd3301915a0777404d0d6627373b979

https://github.com/Ultimaker/Cura/commit/1d6b7c71cbd3301915a0777404d0d6627373b979#diff-a898e447d66fcc2aadbb675de2107611R300
Start and end gcode used to be not-encoded, they are now.

smartavionics, 2017-11-30:
Ah, yes, that has only just been merged. Over to you @Ghostkeeper  and @LipuFei.

fieldOfView, 2017-11-30:
Since I also need my start & end gcodes, I made a fix:
https://github.com/Ultimaker/Cura/issues/2884
Turns out the gcode snippets were being encoded twice.

smartavionics, 2017-11-30:
Thanks a lot for doing that @fieldOfView, is everyone else on holiday?

fieldOfView, 2017-11-30:
I guess they are hard at work on all of our other pull requests...

smartavionics, 2017-11-30:
Ha, that's a good one. My PRs will be suffering from bit-rot before they get around to looking at them!

smartavionics, 2017-11-30:
A question about your fix, what about the other scripts that people can use that are related to tool changes? Are they likely to have the same problem?

ChrisTerBeke, 2017-11-30:
I'll get to your PR's asap! Awesome prints and work guys!

fieldOfView, 2017-11-30:
You mean extruder start/end gcode snippets? Those were broken too, and are fixed with this PR.

smartavionics, 2017-11-30:
Hi @ChrisTerBeke, thanks, you should like the mods I have made to the adaptive layer code.

@fieldOfView, OK, thanks

ChrisTerBeke, 2017-11-30:
@fieldOfView I'll ping Ruben physically now, he made those changes (but should be fixed asap indeed).

closed 2017-11-30 by ChrisTerBeke

smartavionics, 2017-11-30:
Hi @ChrisTerBeke, just reminding you of what I said above: One thing I did notice is that on line 137 of FffPolygonGenerator.cpp you are doing `adaptive_layer_heights->getLayers()` even if adaptive_layer_heights is null.

smartavionics, 2017-11-30:
That still needs to be fixed by you.

ChrisTerBeke, 2017-11-30:
Yes, I actually have that file in front of me right now :)

ChrisTerBeke, 2017-11-30:
https://github.com/Ultimaker/CuraEngine/commit/3e7968163e073fb5e3ec6d3c6dd8f5ca0ca6960f.