Ultimaker/Cura #2879: Cura 3.1 beta - import profiles issue

amagr0, 2017-11-29:
https://ultimaker.com/en/community/52135-ultimaker-cura-31-beta?page=2&sort=#reply-193819

If you need any other file please ask.
Tks

diegopradogesto, 2017-12-01:
This should be fixed in the final release. Thank you for sharing all this information with us.

ChrisTerBeke, 2017-12-01:
This was indeed fixed by 3abf067d25079520d3cf8ddffd216bce5c467cb5 and 7e4ed8a25703416183c1ee3453be2c97d4e4d895.

closed 2017-12-01 by ChrisTerBeke

amagr0, 2017-12-04:
It seems that is not completely fixed.

https://ultimaker.com/en/community/52135-ultimaker-cura-31-beta?page=2&sort=#reply-194149

ChrisTerBeke, 2017-12-04:
The beta was not updated. The final release won't hit the servers until tomorrow...

When installing the stable version, be sure to clean up the folders that were created by the beta.

amagr0, 2017-12-05:
Your colleague Msuurmond sended me new beta to test and confirm that this issue was solved, via forum: https://ultimaker.com/en/community/52135-ultimaker-cura-31-beta?page=2&sort=#reply-193946

amagr0, 2017-12-05:
The stable 3.1 has the same issue.

ChrisTerBeke, 2017-12-05:
Can you upload your profile files so we can double check? Our own profile files don't give any issues, regardless of machine or version they came from.

flugwaps, 2017-12-05:
[fine-raft.zip](https://github.com/Ultimaker/Cura/files/1532690/fine-raft.zip)
Hello, I just updated from 3.0 to 3.1 and wasn't able to import the attached profile.

Hope that will help to fix the bug.

Thx.

amagr0, 2017-12-06:
Hi @ChrisTerBeke, I already sended the [profile ](https://ultimaker.com/en/community/52135-ultimaker-cura-31-beta?page=2#reply-193889) on 29th November to your colleagues.
I will send you again, but the information on the forum need to be deliver to you guys.

https://we.tl/YO9jTGVj2U

Ghostkeeper, 2017-12-19:
Flugwaps' profile imports fine for me. Maybe we fixed this recently in Master?
[image]