Ultimaker/Cura #2881: Cura 3.1beta - hotend temperature issue

caoxiangfish, 2017-11-30:
When I saved the G-code file. The hot-end temperature was always 205° except the initial layer, whatever the value of the "printing temperature" setting is.

ChrisTerBeke, 2017-12-01:
There were some issues in the 3.1 beta with resolving settings. Please remove the user data folder and try again with the 3.1 stable when it comes out next week.

badvision, 2017-12-05:
I was also having the same problem.  It shows the settings correctly at the end of the file but the M104 and M109 commands are also showing an incorrect temp.   Temp should have been 240 but was set to 215

ChrisTerBeke, 2017-12-05:
This is fixed in the stable release today. Please make sure to throw away your beta user data folder in order to get the correct settings upgrade.

closed 2017-12-05 by ChrisTerBeke

vursalov, 2018-01-19:
I founf a woraround. In the configuration settings of printer (change G-code):
M109 S{material_print_temperature} need to set M109 S235 where 235 is a temperature first layer