Ultimaker/Cura #2885: stl file keeps moving to the right

newdudehere, 2017-11-30:
every time i open a STL file, it moves a little more to the right. And now it's completely offscreen.
Is there a simple way to fix this?

diegopradogesto, 2017-12-01:
I don't understand the problem. The model should be automatically placed in the center. Could you provide some more information such as pictures, the STL or the Cura project file that you are using, Cura version, ...
Thanks.

newdudehere, 2017-12-04:
well, the cura version that i use is 3.0.4.
a picture will be of no use, the file is already offscreen. (I don't have an actual 3d printer yet, i'm just testing designs)
and i have no idea how to upload an STL file here, it says github doesn't support it.

Ghostkeeper, 2017-12-15:
You can put it in a .zip file and then upload it to Github.

ianpaschal, 2018-02-28:
Needs info but never got it.

closed 2018-02-28 by ianpaschal