Ultimaker/Cura #2886: Settings Update Very Slowly in Cura 3.0.4

cturley, 2017-11-30:
I'm using Cura 3.0.4 on Xubuntu 17.10 x64. The machine is a Core 2 Duo E4500 (2.2GHz) with 3GB RAM and a Radeon HD3450 graphics card. When I edit printer, material, or profile settings, it may take up to around five seconds for the program to respond. During this time, one of the CPU cores is pegged. I haven't noticed any performance issues besides this. The 3D build plate views and slicing are pretty fast.

I have already tried deleting all of the hidden Cura folders in my home directory and creating new profiles. This seemed to help some. I think before I did this, the lag was a little worse than it is now. I also disabled unused plugins with no effect. Thanks.

ChrisTerBeke, 2017-12-01:
Mm I'm not aware of any changes in applying settings between version 2.7 and 3.0. Maybe you can try the 3.1 beta (or stable 3.1 next week) and see if it still happens for you?

cturley, 2017-12-01:
I just tried 3.1 beta. It seems like materials editing may be faster, but printers and profiles are still slow. I think I may have also experienced this issue in 2.7, although I can't remember for sure.

DanalEstes, 2017-12-11:
Cura 3.0.x and 3.1.0 have been INCREDIBLY slow for me.  Just replacing a two or three digit parameter is painful.  

Wipe cursor over two digits.  Type the replacements.  Several seconds of pause, then the first digit appears.  Up to 20 or 30 seconds of pause, then the second digit may or may not appear. 

If the single digit value would cause a red or yellow field, this often occurs after the 30 second delay, and sometimes the second digit fixes this, sometimes the second digit gets lost. 

This may not be the fastest hardware in the world, but it is not the slowest. The really compelling thing:  Every other piece of 3D Printing software I run, from Fusion 360 to several other slicers, everything runs fine on this machine. And CAD software, such as DesignSparkPCB, Eagle, BOBCad, CamBam, etc.  Everything I do is quite responsive. 

Except Cura. 

[image]

dmitrygorbenko, 2017-12-27:
The same for me.
Next bug https://github.com/Ultimaker/Cura/issues/2771 seems the same as this one, hope developers will take a look at both of them.

alekseisasin, 2018-01-15:
The issue occurs that after each change Cura tries to re-validate all fields and it cost resources. I have  small fix which should reduce small amount of this request, but I do not think it will be enough. To speed validation will require some refactoring as well.

fieldOfView, 2018-01-15:
> The issue occurs that after each change Cura tries to re-validate all fields and it cost resources.

The strange thing though is that on some machines, this seems to take *much* longer than on others. Almost as if something else is also happening.

jackha, 2018-01-22:
I found one problem (and fixed it) that partly explains the speed issues. It did the error checking for all existing extruders, not only for the ones actual in use. For the Custom FDM printer especially there are 8 extruders. I'm currently investigating if there is something more.

devs: CURA-4785

cturley, 2018-01-26:
Tried out Cura 3.2 Beta. Setting changes appear to update much faster. Down to about a second now. Also, Cura appears to startup faster.

DanalEstes, 2018-01-26:
Updated to Beta 3.2.0  AND  deleted any file that looked related to extruder 2 through 8 from all the various directories in C:\Users\Danal\Appdata\Roaming\cura\3.2\...

This has made a large difference in GUI responsiveness.  MUCH more usable.  Launch seems about the same.  Once launched, changing numeric values in the various custom fields has no humanly perceptible lag. 

MUCH MUCH better. 

Also, checked the various log files, and they look much cleaner.  No more repetitive messages about extruders.  Attached them, just for your curiosity. 
[cura.log](https://github.com/Ultimaker/Cura/files/1668045/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/1668043/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/1668044/stdout.log)

DanalEstes, 2018-01-26:
Also... ADDING a new custom printer in 3.2.0 DOES create extruders 1-8, even when the dialog is set to 1. 

(Re-) launching cura with these present is slow, and the gui is slow. 

Deleting this new set, deleting 2 through 8, makes Cura GUI fast again.

jackha, 2018-01-29:
@DanalEstes strange that it's still slow when extruders 2-8 are there! I'll have a look when I have time.

DanalEstes, 2018-02-14:
> Does this still happen in 3.2 stable?

3.2 Beta was better (improved GUI responsiveness).  3.2 Beta with extruder
2 through 8 on custom printers deleted (by hand) from the configuration
directories was MUCH MUCH better, very usable, and quite close to most
other software responsiveness.

I went back to 3.1 due to some other (known and now fixed) beta things... I
will report on 3.2 Stable in the next 48 hours.

Thanks for your help with this.

Danal

ChrisTerBeke, 2018-03-02:
We consider this an ongoing issue, but closing this specific one.

closed 2018-03-02 by ChrisTerBeke