Ultimaker/Cura #2887: can't slice automatically

twitteking, 2017-12-01:
i check the box for autoslice and i restarted CURA, di this many time, but cura still dont slice automatically on startup, i have to go to configure cura and check th ebox everytime!! is there is hotkey/shortcut for slicing? please add if not

also the colors in layered view line type, can you guys use colors with less saturation? those sharp colors hurt my eye!! thank you very much!!

diegopradogesto, 2017-12-01:
It's weird, you shouldn't need to restart Cura to apply that setting and it should be enabled by default. In the cura.cfg file it is a setting that indicates the status of this option. You can check manually if it's correctly set (`auto_slice = True`). In future versions a Ctrl+P shortcut will be added for slicing.

Regarding the colors in line type color scheme, I think they will not change in the near future. By the way, Cura has the material color scheme and, from version 3.1, the feedrate scheme that you can try.

fieldOfView, 2017-12-01:
The color scheme is themeable. You can create your own theme that only changes the layerview colors.
* Go to Help -> Show configuration folder
* In the folder that opens, locate the themes folder
* In that folder, create a new folder named eg "subdued layerview"
* In the "subdued layerview" folder, create a file named "theme.json", containing the following:
```
{
    "metadata": {
        "name": "Subdued layerview",
        "inherits": "cura-light"
    },

    "colors": {
        "layerview_ghost": [32, 32, 32, 96],
        "layerview_none": [255,  255,  255, 255],
        "layerview_inset_0": [128,  0,  0, 255],
        "layerview_inset_x": [0,  128,  0, 255],
        "layerview_skin": [128,  128,  0, 255],
        "layerview_support": [0,  128,  128, 255],
        "layerview_skirt": [0,  128,  128, 255],
        "layerview_infill": [128,  96, 0, 255],
        "layerview_support_infill": [0,  128,  128, 255],
        "layerview_move_combing": [0,  0,  128, 255],
        "layerview_move_retraction": [64,  64,  64, 255],
        "layerview_support_interface": [32, 96, 128, 255],
        "layerview_nozzle": [90, 83, 33, 50]
    }
}
```
* restart Cura and go to Preferences. Select the theme named "Subdued layerview"
* restart Cura again
* check out layerview

Thisismydigitalself, 2017-12-01:
Ah? CNTL+P for slicing?
I use it now for sending to printer using OctoPi...

ChrisTerBeke, 2017-12-01:
Ctrl+P would be a bit weird for slicing, as that's the system default shortcut for any software that supports '2D printing'. I'd expect Ctrl+P to actually print over the selected output device. For slicing we could maybe use Ctrl+M, which is used more often for features like 'submit current line', 'run analysis' or even as duplicate of a carriage return (enter). Coincidentally it could stand for 'make' :P

Ghostkeeper, 2017-12-13:
I can't reproduce the original bug. It seems to always autoslice fine (if the setting is enabled). How can we reproduce this bug? Anything special that you did?

ianpaschal, 2018-02-28:
Needs info and never got it.

closed 2018-02-28 by ianpaschal

Ghostkeeper, 2018-03-23:
By the way, I've found that F5 is a handy shortcut for slicing. F5 reloads the models which then triggers a reslice.