Ultimaker/Cura #2889: 3.1 VersionUpgrade creates broken extruder stack for single extrusion machines

fieldOfView, 2017-12-01:
When upgrading a single extrusion machine instance from 3.0 (global stack only) to 3.1 (global stack + 1 extruder stack), the version upgrade system puts the global "user" container in both the global and the extruder stack. This causes settings that are limited to an extruder not to "take".

steps to reproduce:
* clear out Cura 3.1 profile
* start Cura 3.0 and create a single extrusion machine (eg UM2)
* close Cura 3.0, start Cura 3.1. The configuration will be upgraded
* try to set the Top/Bottom Line Width to 0.11111

expected result:
* setting takes, as any other setting

actual result:
* setting seems to take, cura starts slicing with the setting, but as soon as the field looses focus the setting is put back to the machine default

Inspect the extruder.cfg file (in configuration folder/extruders) created by the versionupgrade step. Note how the top container references the global "user" container, instead of an extruder-specific "user" container.

ChrisTerBeke, 2017-12-01:
@LipuFei didn't we fix this last week?

fieldOfView, 2017-12-01:
Ah, that could be true. My configuration has been upgraded a while ago, and once it has been upgraded to the broken state it will not be upgraded again. Will test.

ChrisTerBeke, 2017-12-01:
That would be great! During QA and regression we found no issue so far (we did a lot of fixes this week for project loading and version upgrading, some which were related to the extruder stack changes).

closed 2017-12-01 by fieldOfView

ChrisTerBeke, 2017-12-01:
Mini heart attack... :P

fieldOfView, 2017-12-01:
My master config was copying from the 3.1 config that was already broken by the 3.1.0 beta.
This is going to need a mention in the release notes for users who have used the beta.

ChrisTerBeke, 2017-12-01:
Good one. @Appesteijn @Vandrasc how to communicate this to beta users?

Appesteijn, 2017-12-01:
I have asked to have this in the blog post that will accompany the 3.1-final