Ultimaker/Cura #2890: Spanish translations broken

thopiekar, 2017-12-01:
```
mkdir -p /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/es_ES/LC_MESSAGES/
mkdir -p /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/fi_FI/LC_MESSAGES/
/usr/bin/msgfmt /<<PKGBUILDDIR>>/resources/i18n/es_ES/cura.po -o /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/es_ES/LC_MESSAGES/cura.mo -f
/usr/bin/msgfmt /<<PKGBUILDDIR>>/resources/i18n/fi_FI/cura.po -o /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/fi_FI/LC_MESSAGES/cura.mo -f
mkdir -p /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/de_DE/LC_MESSAGES/
/usr/bin/msgfmt /<<PKGBUILDDIR>>/resources/i18n/de_DE/fdmextruder.def.json.po -o /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/de_DE/LC_MESSAGES/fdmextruder.def.json.mo -f
mkdir -p /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/de_DE/LC_MESSAGES/
/usr/bin/msgfmt /<<PKGBUILDDIR>>/resources/i18n/de_DE/fdmprinter.def.json.po -o /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/de_DE/LC_MESSAGES/fdmprinter.def.json.mo -f
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
/<<PKGBUILDDIR>>/resources/i18n/es_ES/cura.po:1140: 'msgid' and 'msgstr' entries do not both end with '\n'
/usr/bin/msgfmt: found 1 fatal error
Built target i18n-create-mo-zh_CN
make -f CMakeFiles/i18n-create-mo-ja_JP.dir/build.make CMakeFiles/i18n-create-mo-ja_JP.dir/depend
CMakeFiles/i18n-create-mo-es_ES.dir/build.make:60: recipe for target 'i18n-create-mo-es_ES' failed
make[3]: *** [i18n-create-mo-es_ES] Error 1
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
CMakeFiles/Makefile2:1071: recipe for target 'CMakeFiles/i18n-create-mo-es_ES.dir/all' failed
make[2]: *** [CMakeFiles/i18n-create-mo-es_ES.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[3]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
cd /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /<<PKGBUILDDIR>> /<<PKGBUILDDIR>> /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/CMakeFiles/i18n-create-mo-ja_JP.dir/DependInfo.cmake --color=
mkdir -p /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/fi_FI/LC_MESSAGES/
/usr/bin/msgfmt /<<PKGBUILDDIR>>/resources/i18n/fi_FI/fdmextruder.def.json.po -o /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/fi_FI/LC_MESSAGES/fdmextruder.def.json.mo -f
mkdir -p /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/fi_FI/LC_MESSAGES/
/usr/bin/msgfmt /<<PKGBUILDDIR>>/resources/i18n/fi_FI/fdmprinter.def.json.po -o /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/resources/i18n/fi_FI/LC_MESSAGES/fdmprinter.def.json.mo -f
Scanning dependencies of target i18n-create-mo-ja_JP
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
```
@Ghostkeeper : Ping

smartavionics, 2017-12-01:
Hi, I created a similar issue earlier today about this but it has since been fixed. Try pulling again?

Ghostkeeper, 2017-12-13:
Yes, this was fixed here: https://github.com/Ultimaker/Cura/commit/7068f6c28c3676b6a99f490c34393b5d01d72c00#diff-d2c0631e5adb068a8413c56e62b6ff92

We removed the whitespace from the translation but didn't think of how gettext would react.

closed 2017-12-13 by Ghostkeeper