Ultimaker/Cura #2891: View orientation buttons should not change distance of camera to build plate?

smartavionics, 2017-12-01:
The new view orientation buttons (or whatever they are known as) are quite nice but I don't want to use them because they zoom out too far and make the object tiny. I would much prefer if they didn't alter the distance of the camera from the build plate (or whatever the datum is).

Too far away!

[image]

Thisismydigitalself, 2017-12-01:
If you're talking about Quick camera controls, i second that. 
in addition when you click on any of these and then zoom in, even if the object is selected it will not zoom to center of screen. Also too dark to notice in dark Theme.

Can i ask how to get this working?..
[image]

smartavionics, 2017-12-01:
I'm talking about the little buttons that have recently appeared in top right hand corner of the screen:

[image]

The variable layer height stuff is in the current development source (master branch). I should think that it will most likely appear in version 3.2 but I am not an Ultimaker person so that's just a guess.

Thisismydigitalself, 2017-12-01:
Yes, these are quick camera controls.
On dark Theme they are almost invisible.
[image]

ChrisTerBeke, 2017-12-01:
Adaptive layers will indeed be in 3.2 as experimental mode feature. I've created a new internal ticket for improvements to the view orientation buttons.

ChrisTerBeke, 2017-12-01:
Devs: CURA-4663 (not added to current sprint yet).

cmatthes, 2018-01-13:
I would second @smartavionics note regarding the zooming behavior. 
Maybe not the most left icon but the other four on the right shouldn't zoom out. 
Personally they are currently of no use to me due to the zooming out.

closed 2018-04-15 by smartavionics