Ultimaker/Cura #2893: master branch - setting values "jump back"

smartavionics, 2017-12-01:
I am noticing some new weirdness with the current master branch. I can change a setting, e.g. min layer time from its current value of, say, 5 to 15 and hit enter. The setting field says 15. I press the Prepare button and the print is sliced and the layer view shows the new layers and the button down the bottom briefly gives me the option to save the file. But... immediately, with no action on my part, the min layer time setting field jumps back to its previous value (5) and the button goes back to prepare. It didn't used to behave like this.

Also, I notice that if I change a setting value and press tab the focus moves to the next setting but the changed value is not kept, the setting I just altered goes back to its previous value. Does tab not have the same effect as enter?

ChrisTerBeke, 2017-12-01:
mm that sounds pretty serious, I'll check it out first thing on Monday (don't have a dev environment at home).

fieldOfView, 2017-12-01:
Sounds exactly like what I saw here: https://github.com/Ultimaker/Cura/issues/2889

Try adding a new printer, and see if it still happens with the new printer.

smartavionics, 2017-12-01:
OK, i'll try that tomorrow.

I guess a problem with living on the leading edge (aka master branch) is that upgrading configs, etc. can be a bit hit and miss.

ChrisTerBeke, 2017-12-01:
@fieldOfView thanks for that connection. @smartavionics while devving we always remove our master config from the user data folder so it gets repopulated with the config from the previous version.

smartavionics, 2017-12-02:
I deleted my existing printers and added them afresh. The problem appears to have gone away. Thanks for the help.
@ChrisTerBeke, that sounds like a good idea, unfortunately, I do not have any config files from a previous stable version, I only have my master branch config files. I do, however, have them in a git repo so if they get really screwed, I can go back to a previously working version.
Closing...

closed 2017-12-02 by smartavionics