Ultimaker/Cura #2894: Feature Request: Notification of missing / altered material in sliced object

Thisismydigitalself, 2017-12-02:
Cura is great. All slicing options and plugins are there for letting us slice the best we can. this comes with a price. it happens more and more that i need to switch many times between Solid view and Layer view just to make sure none of the settings effect the outcome in a bad way or even worse - missing entire feature of the soon to be printed object.

So I'm thinking out loud here - Will/Can it be possible to somehow raise a flag when a setting effects that part being printed? I understand that this request is not trivial cause anything can and will raise this flag but I have to ask this as lately playing with experimental plugins which are great, i found out that forgetting a checked setting might be left unnoticed and it will completely change the part and remove features without ever noticing it - the only way to know that is to scan the object from all angles while switching between Solid and Layer views back and forth many times.

A penny for your thoughts.

Ghostkeeper, 2017-12-14:
Myself, I just stay in layer view most of the time.

It would be much easier to see what changes if we didn't delete the layer view data while a slice is going on. This was decided upon early on by our UX designer though to prevent confusion: One might think that the slice has already completed and you're looking at the result, while it's still slicing and you're actually looking at the old layer data from your previous slice.

Actually indicating what lines changed from the previous slice is computationally very expensive. You'd need to put all old lines in a giant hash set and then for each line in the new g-code test whether that line is already in the set. You'd need to implement this in native code (likely C++) because of performance, and what's more you're going to have to keep track of twice as much layer data, and more because it's in a very memory inefficient format. It's technically possible, but I don't see that happening any time soon.

Per part indicating which settings affect that part is a bit weird. Many settings affect every part. Nearly all settings in the Quality, Shell, Infill, Material, Speed, Travel and Cooling categories will affect every part that has volume. I'm not quite sure what you had in mind. Maybe I understood your question wrong.

The problem of accidentally having walls set to 0 or something should be handled in some way by the warning values. We colour the setting orange if we think it's probably not going to print well. We can't really detect cases of the infill being disabled though because that could very well be what you want.

Thisismydigitalself, 2017-12-14:
Hello there and thanks for the conversation.
To begin with, i completely understand the challenge with what i'm asking - if possible at all.
What triggered my question was when out of so many settings there are,  i once left the "Remove all holes" checked and noticed the missing hole just before printing. you can not see a missing hole in layer view unless you know your model that well that you can't miss it.  the holes will be removed from "Layer view" but will be seen in "Solid view". this is one example out of many of what should not be happening. we must have a clear way of knowing a feature was just removed and raise a flag. maybe not comparing between Solid and Layer view, but maybe have Cura UX let the user know a setting which alters the model core features is ON. maybe a 50% dissolve between Layer and solid views?.. I don't know.

I hope my point makes sense. 
There are so many settings these days which are there to help us, but sometimes its like looking for a needle(s) in a haystack. it would be nice if Cura had a "manometer" to worn us they're there.

fieldOfView, 2017-12-14:
> This was decided upon early on by our UX designer though to prevent confusion

And I stand by that decision. Although saying that, perhaps a "ghost" of the previous layerview could remain until new layerdata is available (eg 50% transparent, or with 50% less saturated colors). It just needs to be painfully obvious that there is something "up" with the current layer view.

Ghostkeeper, 2018-01-02:
Maybe render it with the striped shader?

ianpaschal, 2018-03-05:
If we strip this down to showing a faded layer view and only clearing/redrawing when a new slice is complete, this is not such a difficult change, right?

Ghostkeeper, 2018-03-29:
No, it's not really hard. It needs another render pass to coalesce the layer view render on a quad and render that quad transparently, because you don't want to render layer view itself transparently. Our rendering system is designed to be able to do that sort of thing. And then to change the signal of layer view to erase its data at a different time.