Ultimaker/Cura #2895: Variable layer height settings

smartavionics, 2017-12-02:
Having played with this new feature a little, I think it would be better if instead of specifying a max variation from the base layer height, one could specify the min and max layer thicknesses (obviously with the contstraint that max layer thickness > min layer thickness). I know that the end result would be the same as what we have now but, to me, it's easier to think of a min and max layer thickness rather than the middle thickness +/- something.

smartavionics, 2017-12-02:
Another thought: in the future, the gui could have a slider with two bugs to set the min and max layer heights?

smartavionics, 2017-12-02:
BTW Chris, why do you have this in calculateLayers() ?
````
    // compensate first layer thickness depending on slicing mode
    if (slicing_tolerance == SlicingTolerance::MIDDLE)
    {
        z_level += this->initial_layer_height / 2;
        this->initial_layer_height += this->initial_layer_height / 2;
    }
````

ChrisTerBeke, 2017-12-10:
To keep the slicing mode offset behavior consistent with non-adaptive layer height mode. Not sure yet if I wanna keep it in though...

Fronscavenger, 2018-01-18:
Sorry for the ignorance of my comment. Why not just apply a variation of layer height to the support?

smartavionics, 2018-01-18:
Hi - being able to print a support layer that is thinner than the layer used for the model would (IMHO) be very useful because at the moment, the gap between the top of the support and the model has to be an integer multiple of the (fixed) layer height and may not be the ideal size.

ChrisTerBeke, 2018-01-18:
Thanks for the suggestions!

Even though that is a good idea, that would require way more architecture changes in the software. The current implementation is the simplest possible (define layer heights for an entire layer at the first stage of slicing). Anything else would be difficult to implement in the current engine structure. The biggest issue is that after creating the layers, everything is seen as a 2D problem by all other algorithms, as those are an order of magnitude simpler to comprehend than 3D problems.

We have a pretty long wish-list for things to add to variable/adaptive layers, but that simply requires a lot more effort to implement (which needs approval from people higher up the food chain at Ultimaker :P ). Let's first do this proof of concept in the new 3.2 release and see where it goes from there :).

doomsed, 2018-01-24:
I agree that "min/max"  is easier to conceptualize than "middle +/- variance". Especially since the "middle" may not have a grounding in physical reality, depending on other settings (if the variance is not evenly divisible by the step size then no layers will have a height of "middle").

since the layer height stepping is already relative to the maximum of the current calculated range, you could use [Layer Height] as the default maximum layer height for adaptive layers and replace [adaptive layer maximum variation] with [minimum adaptive layer height]. 

It might also be nice if there was a tag or notification on the [step size] and  [maximum variation] or [minimum layer height] that the total range may not be used whenever {remainder(2 x [maximum variation] / [step size]) !=0} or {remainder( ([max height] - [min height]) / [step size]) !=0}

It would also be easier to conceptualize [adaptive layers threshold] if a percentage of [line width] was used. i.e. "The layer height will decrease if less than x% of the outer wall width overlaps with the outer wall of the previous layer"

noisemaker00, 2018-02-15:
I agree, in my opinion is better to replace current UI settings with:

- Max layer height
- Min layer height

Is more user friendly and by now, using the delta (+/-) the maximum height may be greater than the printer nozzle is able to achieve.

In addiction to this, please see: #3324

slikk66, 2018-02-28:
Agree about +/- used this for first time today. I'd like .12 min, .2 max.. Took me a bit to realize I needed to set my layer height to .16 since that's not a number I was intending to use, just doesn't seem to pass the "makes sense" test.

slikk66, 2018-02-28:
Also, to make this even easier to patch in, why not have the current height setting become the minimum, and allow you to set an upper bound for max with the step setting. Internally split the difference and set that to the "layer height and apply the same logic as is current.

Ghostkeeper, 2018-02-28:
Our project manager removed this feature from our planning. We won't be addressing it soon. Maybe later.

doomsed, 2018-03-02:
Removed this topic from planning or removed Adaptive Layers from planning?

ChrisTerBeke, 2018-03-02:
Removed improvements for adaptive layers. Although it's back on the planning now, just later this year.

ianpaschal, 2018-03-02:
@doomsed This improvement idea. Adaptive layers are still very much something we're working on. The `Unplanned` label denotes something which we currently won't do but may very well in the future (like adding improvements to an experimental feature), versus `Won't Fix/Do" which denotes things we know for sure just ain't gonna happen (like fixing bugs in version 2.4).

Ghostkeeper, 2020-08-12:
With new insights explained [here](https://github.com/Ultimaker/Cura/issues/8174) we've decided to reject this feature request.

closed 2020-08-12 by Ghostkeeper