Ultimaker/Cura #2897: The Start Gcode for Creality CR-10 is faulty

markosiponen, 2017-12-03:
The Start Gcode for Creality CR-10 is faulty. 10mm filament is extruded by the Start Gcode. But the extrusion distance is not reset at the end of the start gcode. You should add a G92 E0 at the end of the Start Gcode to fix this.
```
G21 ;metric values
G90 ;absolute Positioning
G28 ; home all axes
G1 Z5 F3000 ; lift
G1 X20 Y2 F1500 ; avoid binder clips
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X120 E10 F600 ; prime nozzle
G1 X150 F5000 ; quick wipe
*** Add G92 E0 here ***
```

nallath, 2017-12-05:
Could you make a pull request for this?

Ghostkeeper, 2017-12-18:
You are using an older version. We recently removed this specialised starting g-code for CR-10: https://github.com/Ultimaker/Cura/commit/73e809b956ce79629cc70d47c94a4e8d3b14ee63

closed 2017-12-18 by Ghostkeeper