Ultimaker/Cura #2898: auto-arrange now fails with tiny model

smartavionics, 2017-12-03:
I have a part that is approx 3mm square. It gets printed in batches and so I load a single model onto the build plate and multiply it. The current Cura master branch fails to auto-arrange this part and so all of the
copies end up in the same place! Larger models (with a footprint approx 6x9mm) do auto-arrange OK.

Here's a pic of the model after it has been multiplied by 11 and sliced:

[image]

alekseisasin, 2017-12-04:
Hi @smartavionics, I tried to do same steps I have another results; I have two simple models 3mmx3mmx3mm and 2,5mmx2,5mmx2,5mm squares and they auto arranged differently. Can you upload your specific model, it depends on the shape of model 
[image]
[image]

alekseisasin, 2017-12-04:
Now I see, actually it happens also with my small models. I didn't have what you described, because my setting "Ensure models are kept apart" is enabled., you can check it should help. For now it can be a fast fix to avoid all models in the same place. 
Devs: CURA-4672

smartavionics, 2017-12-04:
Hi, I tried enabling that and it made no difference for me.

smartavionics, 2017-12-04:
Ah, you have to restart Cura for the ensure models are kept apart setting to be effective. Now, it is arranging the little guys. They are still very close together. It would be better if you could arrange models without having to have the keep apart setting enabled. Also, when I used a slightly bigger model, it did arrange them OK without the keep apart setting being enabled.

smartavionics, 2018-01-15:
This appears to have been fixed, thanks. Closing...

closed 2018-01-15 by smartavionics