Ultimaker/Cura #2899: Re-Slicing with "Open ..." Dialog

thesec, 2017-12-03:
Whenever I select file/open... the current model is automatically being sliced again.
Without any setting having changed at that moment that should not happen.

ChrisTerBeke, 2017-12-03:
Indeed. I think it also happens when you connect to a networked printer.

smartavionics, 2017-12-03:
Hi @ChrisTerBeke , maybe related to the above issue, updating (aka saving) the current profile also triggers a reslice (well it would do if I had auto-slice enabled, you know what I mean).

fieldOfView, 2017-12-03:
Add "saving a project" to the list...

ChrisTerBeke, 2017-12-03:
Sounds like this needs a Jira ticket. My gut says it has to do with one or more of the many signals being thrown around in Python to which re-slice reacts.

edugarci, 2017-12-04:
hola buenas noches,tengo un grave problema, se me a perdido la comunicación con mi impresora prusa i3
y lo e intentado casi todo sin lograr conectarme, no reconoce el puerto. haber si me podeis ayudar porfa

Ghostkeeper, 2017-12-15:
I can't reproduce this. I can reproduce the bugs that a reslice happens when you save your profile or save a project file (we already had tickets for those). No reslice happens when you connect to a networked printer though, unless you sync the materials in the subsequent dialogue but then it's necessary. Also no reslice happens when I click on File -> Open File... until I actually load a new model, but then it's also necessary.

Ghostkeeper, 2017-12-15:
Edugarci, your problem is a different one than what is reported here. I would like you to make a new issue.

Some hints I can give with your connection problem: Try making a connection with Pronterface first to see if it's Cura or your printer itself that is giving the problem.

closed 2018-11-26 by Appesteijn