Ultimaker/Cura #2900: Allow for a PrimeObject instead of a PrimeTower

MB3Y, 2017-12-03:
I know this is not a bug, but a feature. Please convert appropriately if possible.
As most people will agree, the prime tower is waste material. But then again, I do have no objections for some objects to look fuzzy multicolored (e.g. if I plan to postprocess and paint to name one)

It would be great to print one object - with a nozzle cleaning object aside. One just has to make sure that the layer surface of that secondary object is sufficiently big to clean out a color swap (IF any does actually occur on this layer!)
Wipping the nozzle on the exterior of the tower would not be possible, but on an oozing shield or maybe even at the interior of this 'waste' object could be an option (or disable these things when it's really an object used as prime tower)

Sounds doable and an asset?
It sure does to me.

BagelOrb, 2017-12-06:
You can just set the layer start position to where that object is...

MB3Y, 2017-12-06:
Hi Tim, very cryptic description... Can I print a MulticolorCat.stl and put a Dog.Stl next to it to serve as PrimeTower such that Cura really does the extruder switches and all the other PrimeTower magic on the Dog?
Mikael

Ghostkeeper, 2017-12-19:
What BagelOrb meant is that you could just print without a prime tower. You can then set the Layer Start X and Layer Start Y coordinates close to the object you want to use as a prime object.

It should start with the infill by default (using most printers/profiles). If not, make sure that the "Infill Before Walls" setting is enabled.

MB3Y, 2017-12-19:
Yezzz, thank you!, that's what I figured AND excercised meanwhile, you do miss out the specific features of a PrimeTower though. For now good enough!

closed 2017-12-19 by MB3Y