Ultimaker/Cura #2901: Prime tower doesn't provide support for more than 2 colors per layer

MB3Y, 2017-12-03:
Cura allows to configure multiple extruders - in my case 3, since I'm using a diamond mixing hotend.
But unfortunately the prime tower only accommodates for 2 colors per layer, not more :(

ChrisTerBeke, 2017-12-03:
You can enable to use a prime tower per used extruder. That means a bit more tower volume, but not bounded to 2 extruders per layer.

MB3Y, 2017-12-03:
Hi Chris,

in my setup there are  several settings that are marked with a "linked" symbol. The comment that appears when hoovering over is "This setting is always shared between all extruders. changing it here will change the value for all extruders".
I assumed that this is also true for the Prime Tower since this symbol also appears next to the option "Enable Prime Tower".
Your comment made me pay more attention and I see there is an addendum "The value is resolved from per-extruder values [false,false,false]" (btw: this addendum is only there when the option is unchecked)

Do I need to get some "...[true,true,true]" before I can execute what you indicate above??
Can you tell me how I can do that please?

Thank you,
Mikael

ChrisTerBeke, 2017-12-03:
Resolvement is an internal system in Cura that sets a setting based on the values of all extruders. We can set rules per setting to get the right value (for example the highest bed adhesion temperature). So even though the settings per extruder can be different, some settings can only have 1 instance of them (again, like bed adhesion temperature) and for those we have these rules set up. You can find all of these rules in the `fmdprinter.json` definition in the `resources/definitions` folder (the `resolve` field is there for some settings).

As for the prime tower settings, I was confused with another slicer I worked on before. Cura does not have a 'tower per extruder' feature. Since that's a useful feature I'll create an internal ticket for it, but I can't say anything about how long it will take to be implemented.

ChrisTerBeke, 2017-12-03:
Devs: CURA-4666 (not added to sprint).

Ghostkeeper, 2017-12-15:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/1936 and https://github.com/Ultimaker/CuraEngine/issues/588 (the latter of which might have been using the same diamond extruder).

closed 2017-12-15 by Ghostkeeper