Ultimaker/Cura #2902: Incorrect documentation of Cura preferences location

stepjanssen, 2017-12-03:
The location of Cura preferences on Mac should be `~/.cura/`, according to this documentation:

https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

However, on my installation, it appears to be:
`~/Library/Application Support/Cura/`

I think the documentation should be fixed, or maybe I'm just using an old version, but based on the change date of the documentation I can't imagine that that's the issue.

fieldOfView, 2017-12-03:
You are right, the documentation needs updating. Fortunately, it's a wiki. So you can update it too!

ChrisTerBeke, 2017-12-03:
Went ahead and updated it for you. Also added some missing info on config stored of older versions.

closed 2017-12-03 by ChrisTerBeke