Ultimaker/Cura #2906: Start and End GCode not as start and end code

MB3Y, 2017-12-04:
Hi,
I'd like my start code to move the head to a position where it can drool, drop and drip while heating up, rather than staying in the center of my eliptical buildplate (home position).
Unfortunately Cura adds the commands to wait for warming up in front of my Start GCode, so heating up always is forced in home position.

Some remark for the End GCode although I'm not hampered here to move the head away from the object while cooling down: Cura still adds commands (switch off temps) after my End GCode block.

Surely I could print all my objects away from the center... but the center position is the default loading spot and also my preferred print position ;)
Cheerio,
Mikael

fieldOfView, 2017-12-05:
Cura only adds its own heatup-sequence if it detects there is no heating up in your start gcode. If you place the following snippet in your start gcode, you have full control over when it happens.

```
M140 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
```

MB3Y, 2017-12-05:
Nice! Works!
Still some questions though:
1. Shouldn't we be using M190 and M109 in a start script? Such that we also wait for the temp to be set before continuing...
2. if you comment these commands out (add ";" in front), the Cura still doesn't generate itself the commands (oops)
3. were to find a list of all possible parameters that Cura recognizes and substitutes in start/end scripts?
Thank you!

fieldOfView, 2017-12-05:
1. Either M109 or M104 work. Actually, Cura checks for the use of {material_print_temperature} or {material_print_temperature}, not the actual M-code (or indeed if the line is commented out)
2. This is sort of by design, I guess
3. See fdmprinter.def.json in resources/definitions

Ghostkeeper, 2017-12-18:
This is indeed the intended behaviour. We want to add the heating commands there if the start g-code doesn't do that itself to make the start g-code less error prone.

closed 2017-12-18 by Ghostkeeper