Ultimaker/Cura #2907: NO reconoce puerto COM3 con Prusa i3

edugarci, 2017-12-04:
hola buenas noches, tengo un problema grave, perdí la comunicación con mi impresora Prusa i3
y lo intente con  casi todo sin lograrlo, no reconoce el puerto. haber si me podéis ayudar porfa.
Borré los puertos para que me la reconociera automáticamente como ultimo recurso y sigue sin comunicarse con el ordenador

ChrisTerBeke, 2017-12-05:
Please ask you question in English?

edugarci, 2017-12-05:
hello good night, I have a serious problem, I lost communication with my Prusa i3 printer
and try with almost everything without achieving it, does not recognize the port. have if I like to help porfa.
Delete the ports so that I am automatically alerted as a last resort and still do not communicate with the computer

ChrisTerBeke, 2017-12-05:
Does it work with other USB printing software? If not, I'm afraid we can't really do anything in Cura to fix it. If it does, it might be that Cura doesn't support the Prusa's type USB connection, but I've seen others use it without issues though.

edugarci, 2017-12-06:
Si, la placa arduino la reconoce, pero no la de la prusa I que es la V1. 2

edugarci, 2017-12-06:
Yes, with the arduino plate alone, but with the prusa i3 do not, which has the plate V1. 2

El 6 dic. 2017 14:07, EDUARDO GARCIA <electrodigit@msn.com> escribió:

ChrisTerBeke, 2017-12-14:
Then it's not really a Cura issue but a firmware/board issue with the printer itself. I'll close this thread then.

closed 2017-12-14 by ChrisTerBeke

Ghostkeeper, 2018-01-02:
Other people seem to work with the Prusa i3 boards fine though. I've seen it work.