Ultimaker/Cura #2910: Machine prefix only one letter in Cura 3.0.4

hoegge, 2017-12-05:
When I save my gcode file only the first letter of the selected printer is added as prefix. E.g. machine name `CR-10` or `A31` becomes `A_Tool.gcode` and `C_Tool.gcode` respectively.

Isn't that a bug or is it another setting?

CURA 3.0.4
Windows 10

smartavionics, 2017-12-05:
Interesting because lately my machine prefix has increased to 3 characters from 2! Kossel XL is now KXL rather than KX?

Ghostkeeper, 2017-12-18:
The abbreviation is generated like this:
* Split the machine name on spaces, so split it into words.
* For every word:
  * Accents are stripped (`â` to `a` and such).
  * If it's a number or at most 3 characters, use the whole word.
  * If it's longer than 3 characters, take just the first character.
  * Uppercase everything.

So `CR-10` is just one word (there are no spaces). It's longer than 3 characters so the abbreviation becomes `C`.

`Kossel XL` are two words. The first word is longer than 3 characters so it becomes just `K` and the `XL` is only 2 characters so that gets pasted after it directly. The result is `KXL`.

For `A31` I would expect it to become `A31` rather than just `A`.

I think we might also have to split upon the dash character `-`, for cases like `CR-10`.

Ghostkeeper, 2017-12-18:
I made a Custom FDM Printer called `A31` to test. My test print job indeed becomes `A31_Cube`, so I can't reproduce that bug.

closed 2017-12-18 by Ghostkeeper

Ghostkeeper, 2017-12-18:
I've made a minor change that causes dashes such as in `CR-10` to be marked as word boundaries.

commit d5b3f18d (2017-12-18): Split all non-word characters for making abbreviations

So now cases like CR-10 are more appropriately split into ['CR', '10'].

Fixes #2910.