Ultimaker/Cura #2912: 3.1.0 does not take over Printers and Profiles (?) from 3.0.4, Kubuntu 17.10

mpsdskd, 2017-12-05:
Today I followed curas advice to update to 3.1.0, so I downloaded  it from ultimaker.com.
Upon opening 3.1 asks me to set up a printer, although I have it all set up in 3.0.4 (which still works, just opening the old appimage), so no printer settings are take over from the previous version. 
Clicking Settings -> Profiles shows a blank window [-> Screenshot] and cura closes itself upon closing the window.
[image]
The same behavior occurs on my other machine, also Kubuntu 17.10.

Kubuntu 17.10, Appimage Cura 3.1.0 from ultimaker.com
- [x] I have searched open and closed issues for duplicates

LipuFei, 2017-12-06:
Hi @mpsdskd, thanks for the zip file! I think this is enough for us to investigate. We will let you know when it is fixed.

mpsdskd, 2017-12-06:
I tried to solve the problem myself in a rather low tech way: 
- remove: /home/*username*/.local/share/cura/3.1/
- copy  /home/*username*/.local/share/cura/3.0/ to  /home/*username*/.local/share/cura/3.1/

Looks like everything works for now....?

LipuFei, 2017-12-06:
Hi @mpsdskd I have fixed this problem in the master branch. For now you can use this workaround : make sure that you have a machine activated (selected) before going to the "Preferences -> Profile" dialog.

Sorry for the inconvenience.

mpsdskd, 2017-12-06:
Okay, I have my other machine to try, but upon opening 3.1 I have no printer to select (which is the main problem). 
So you want me to create a new printer?
I did that and now there is no problem opening the profile settings.

But my main problem is, that my old printer is gone in the new version.

LipuFei, 2017-12-06:
Ah ok. I will have a look. I used your cura directory to upgrade and I could find all the printers in the top-right list, although none is selected when Cura first starts, so I thought it was just a problem with opening that dialog.

[image]

mpsdskd, 2017-12-06:
Interesting, for me there is only the newly created ultimaker 3.
Did you try it on Linux?

LipuFei, 2017-12-06:
On Windows 10. Good point, I will try it on Linux.

LipuFei, 2017-12-06:
You are right. It's not working on Linux somehow.

mpsdskd, 2017-12-06:
Can you confirm that my work around works for you as well?

- remove: /home/username/.local/share/cura/3.1/
- copy /home/username/.local/share/cura/3.0/ to /home/username/.local/share/cura/3.1/

LipuFei, 2017-12-06:
@mpsdskd There is a problem in copying files over during the upgrade. I suggest to do the following two steps and it should work just fine:
- remove `~/.config/cura/3.1`
- remove `~/.local/share/cura/3.1`

In case you are interested, I have made a fix: https://github.com/Ultimaker/Uranium/commit/0fbb3a147a6f13ff32920079c48f260d8fe7679d

mpsdskd, 2017-12-06:
Does not fix my problem:
I removed ~/.config/cura/3.1 and ~/.local/share/cura/3.1 and started 3.1 again.
I get asked to create a new printer and my old printers and profiles are gone.

LipuFei, 2017-12-06:
Have you also removed `~/.config/cura/3.1`?

mpsdskd, 2017-12-06:
Yes

mpsdskd, 2017-12-06:
However I did not copy ~/.local/share/cura/3.0 to 3.1 as you did not tell me to.
Did you want me to do this as well?

LipuFei, 2017-12-06:
That's weird then... I will check later. Thanks

DanalEstes, 2017-12-11:
Slightly different symptom, so this may or may not help debug:

I converted from 3.0.4 to 3.1.0, via the "Ultimaker Cura-3.1.0-win64.exe" installer downloaded 12/10/2017.

All printers came forward.  All materials came forward. All customized print profiles did not. 

Windows 10 Home, V1703, Build 15063.726.

[cura.zip](https://github.com/Ultimaker/Cura/files/1548266/cura.zip)

LipuFei, 2017-12-13:
Hi @mpsdskd , I will close this issue if you don't mind since it's working for you now.

mpsdskd, 2017-12-13:
Okay, I hope the next update will be easier

closed 2017-12-13 by mpsdskd