Ultimaker/Cura #2915: Cura 3.1 crashes when slicing

Bobsy5361, 2017-12-05:
when slicing stl. Cura crashes towards the end of the slice

ChrisTerBeke, 2017-12-05:
We need more info. Can you post the log file and models/settings?

Bobsy5361, 2017-12-05:
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/1532558/cura.cfg.txt)

Bobsy5361, 2017-12-05:
added last part of cfg from loading. full cfg to big to be added to this issues box.. for info under Cura 2.3.1 stl loaded / sliced and printed correctly.

Bobsy5361, 2017-12-05:
[image]
scale up to 205%

ChrisTerBeke, 2017-12-10:
Relevant lines:

```
2017-12-05 14:29:50,650 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [609]: Backend quit with return code 3221225477. Resetting process and socket.
2017-12-05 14:29:50,876 - INFO - UM.Backend.Backend._onSocketError [202]: Backend crashed or closed.
```

Not sure yet what's going on though, must be something with this particular model. Could you upload the 3D files?

ChrisTerBeke, 2017-12-10:
Also seeing this:

```
2017-12-05 14:30:51,307 - DEBUG - UM.Backend.Backend._onSocketError [200]: Socket was unable to bind to port, increasing port number to 49675
2017-12-05 14:30:51,309 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-12-05 14:30:51,311 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [232]: Attempting to kill the engine process
2017-12-05 14:30:51,313 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-12-05 14:30:51,415 - DEBUG - UM.Backend.Backend._onSocketError [200]: Socket was unable to bind to port, increasing port number to 49676
2017-12-05 14:30:51,418 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-12-05 14:30:51,420 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [232]: Attempting to kill the engine process
2017-12-05 14:30:51,422 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-12-05 14:30:51,527 - DEBUG - UM.Backend.Backend._onSocketError [200]: Socket was unable to bind to port, increasing port number to 49677
2017-12-05 14:30:51,529 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-12-05 14:30:51,531 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [232]: Attempting to kill the engine process
2017-12-05 14:30:51,532 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
```

Somehow all socket ports that CuraEngine is running on are already in use.

Bobsy5361, 2017-12-11:
Hi ChrisThere have been several files that the program has crashed on even a simple cube 75 x 75 x 75 mm 15% infill.. It's always as it gets toward 80 to 90% of the splicing completed.RE: the sockets inuse is that looking at ports? as I always print from sd.The one strange thing is that i am using 2.3.1 vers and it's all fine?Can I delete the old log and start a new one.. this one is + 10mb.
I prefer Cura to simplify3d, I get far better results also for info in the experimental i am using the draft shield for ABS. It works really well. I was unable to print anything ABS before without warping or layer splitting. The only thing i see is the layers of the shield are not wleding to each other all the time and find i can split the sheild along the layers every 10mm of the height. Maybe thats it doing its job and stopping the cooling on the inside.Can you please add the facility to create and delete your own supports as in simplify3d.. This is an excellent option and save a lot of filament also to be able to add muliple temps at different layers..as yet again in simplify.
Best wishes for christmas to the team excellent program 
Bob

Bobsy5361, 2017-12-11:
CHRISTERBEKE
Spoke to soon after email Chris. I can't add the stl but the cube caused a crash on splice.
[cura.log last print splice.txt](https://github.com/Ultimaker/Cura/files/1549485/cura.log.last.print.splice.txt)

[image]

Bob

LipuFei, 2017-12-13:
@Bobsy5361 Could you save and share it as a project file? via menu "File -> Save project". Thanks.

Chabs13, 2017-12-23:
I had the same problem on 3.1, fixed it by slightly lowering the z axis of the model.  May have something to due with 3.1 allowing floating models.

Ghostkeeper, 2018-01-15:
Also saw in that log that there were still references to `{travel_speed}` (again). Fixed those up here: https://github.com/Ultimaker/Cura/commit/e6cd6c264590d0dac42980cec6dc61f3f2346c98

Ghostkeeper, 2018-01-15:
In your logs it looks like Cura's front-end crashes at some point?

In any case, I can't reproduce it with a normal cube, even one that floats in the air (with drop-to-buildplate disabled).

ChrisTerBeke, 2018-05-23:
No new info, plus can't reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke