Ultimaker/Cura #2916: Cura 3.1 is not launching on Win 10

AJanosch, 2017-12-05:
Hi,
I have updated to Cura 3.1 today, (uninstalling Cura 3.0)

double-Clicking Cura 3.1 the first startup screen is displayed, but then it is not launching
Cura is also not seen in windows task manager. 

Any Ideas ?
is there any Log-file  - for further debugging?
Thanks

modem7, 2017-12-05:
Windows 7 as well here. Gets to the loading machines, splash screen disappears, then nothing. Once splash screen goes, Cura does not show up on task manager

LipuFei, 2017-12-06:
Hi @AJanosch Could you get the log file? It's located at `C:\Users\<your-username>\AppData\Roaming\cura\3.1\cura.log`.

AJanosch, 2017-12-06:
There is no file in Cura Directory in C:\Users\<your-username>\AppData\Roaming\.

all I found is this the following files concerning Cura 3.1 in C:\Users\<my-username>\AppData\Local\cura

[stderr.log](https://github.com/Ultimaker/Cura/files/1534785/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/1534786/stdout.log)

fieldOfView, 2017-12-06:
For clarity: you edited those files to replace your username, right?

Does ```\\storage-02\SYS_UserProfiles\<MyUsername>\AppData\Roaming\cura\3.1\machine_instances\Ultimaker+3+%232.global.cfg``` exist?

AJanosch, 2017-12-06:
yes, I replaced my Username

this opens:

\\storage-02\SYS_UserProfiles\MyEditedName\AppData\Roaming\cura

AJanosch, 2017-12-06:
Okay, there I found a Cura 3.1 directory:

here is the cura.log :

[cura.log](https://github.com/Ultimaker/Cura/files/1534969/cura.log)

LipuFei, 2017-12-06:
Hi @AJanosch , thanks for the log. Looks like something went wrong during the upgrade. Could you package your `cura` directory (in ZIP or similar) and attach it here? That will help us identifying the problem. Thank you!

AJanosch, 2017-12-06:
I will do-
could you send please send me a non public upload option /  e-mail address option

LipuFei, 2017-12-06:
l.fei@ultimaker.com

AJanosch, 2017-12-06:
An e-mail is on the way...

modem7, 2017-12-06:
Sent my logs as well so you get a windows 7 log also

LipuFei, 2017-12-07:
Hi @modem7 , I have emailed you a set of manual instructions. Hopeful that will fix your problem.
@AJanosch I'm still checking your case. It seems to have something to do with those (network) file paths.

LipuFei, 2017-12-07:
Hi @AJanosch, I fix this problem in master: https://github.com/Ultimaker/Uranium/commit/a2e205e78d644ad633ae608c5f1ce9e8520f1ff3
This is caused by the extra dots in your path judging from your log file: `\\storage-02\SYS_UserProfiles\<MyUsername>\AppData\Roaming\cura\3.1\./machine_instances\.\Ultimaker+3+%232.global.cfg`
The extra dots caused the file moving function `os.remove()` to fail.

Unfortunately, I don't have a good workaround for you at the moment. Sorry for that. I will discuss with the team see what we can do for you so you can use Cura 3.1.

AJanosch, 2017-12-07:
Hi LipuFei,
I will discuss this with my SysAdmin as well.
(for me it seems to be strange that files that IMO should be on a local  machine are  redirected to a server volume. But maybe this is how it is working in the (our) current Windows universe???
Thanks - I will also try to find a work around (like installig it locally - without being connected to the a Microsoft domain).
Thanks - I will keep you updated.

LipuFei, 2017-12-07:
@AJanosch It is actually not so strange. Some organizations do store those accounts on a central server so whenever you log in, everything will be synced. Also it is easy to manage those accounts and disk quota. For now, if you can use a local account, it should work for you.

ChrisTerBeke, 2017-12-10:
Related to #2798?

fieldOfView, 2017-12-10:
No, not related to #2798

ChrisTerBeke, 2017-12-10:
Seeing different errors in the log indeed.

AJanosch, 2017-12-11:
Just a feedback - installing Cura 3.1 locally (doesn´t work either) on my Win 10 machine.

And a (stupid) question - how could I try your fix (in Ultimaker/Uranium) 

Thanks

LipuFei, 2017-12-11:
Hi @AJanosch, are you using a local user on that Win10 machine? and Could you post the log file on your Windows 10 machine?
I currently don't have access to the build server. My colleagues will make a new installer with the fix and send it to you.

Ellecross, 2017-12-11:
Hey @AJanosch , we've a new build for you to try. Lipu has applied a fix and I've uploaded the installer. You can download it here: https://www.amazon.com/clouddrive/share/xCdJIxHQ9CRiL69N6ImnkuLEaa2oht7TqmxG0ykCIIj Can you try this and see if your issue will be resolved?

AJanosch, 2017-12-11:
Unfortunately not resolved - sent the log to LipuFei
(Install was without deleting the old version of Cura)

AJanosch, 2017-12-12:
Perfect - thanks a lot @LipuFei Fei & dear Ultimaker team.
this was a hard one - but you did everything possible & replied promptly & your Best 
Cura 3.1 (the new Beta) is working - like a charm.
Thank you so much . 
Perfect Support!!!!

closed 2017-12-12 by AJanosch