Ultimaker/Cura #2917: Cura3.1 release: Mirror bug

Thisismydigitalself, 2017-12-05:
Before mirroring
[image]

After mirroring  (does not matter on which axis)
[image]

Thisismydigitalself, 2017-12-05:
Apparently more issues..
the drill hole you see at the bottom is not a cut through all the way to the other side as it should be.
Downgraded to 3.0.4 and both issues are gone.

Will stay with 3.0.4 for now.

Thisismydigitalself, 2017-12-07:
It would be nice to know you have managed to reproduce these artifacts.
I can. with any stl.

ChrisTerBeke, 2017-12-07:
We're having a cramped week. Someone will pick this up either tomorrow or early next week. Sorry for the delay.

smartavionics, 2017-12-10:
Hi, could you please make that stl available so I can try and reproduce this problem, thanks.

Thisismydigitalself, 2017-12-10:
Sorry, can't share this specific stl but on 3.1 with any stl introduces the same issues. 
As far as i can see on 3.0.4 mirroring works OK for me.

ChrisTerBeke, 2017-12-10:
Could you compare the command line instructions that are send to CuraEngine between versions with the same settings? That might give us an indication of which settings are actually differently sent to or differently executed by the engine.

Ghostkeeper, 2017-12-18:
It looks a bit as if you enabled the experimental setting "Make Overhang Printable".

Thisismydigitalself, 2017-12-19:
Just tried this "Make Overhang Printable" on a different part - It does looks so but then again - how do you explain the effects only on the mirrored copy?  i am trying to locate the project i saved to know for sure. This just made me think that if one saves a project with two parts, one which is mirrored, you do not have means to know which was mirrored and which was not unless i am missing something.

Ghostkeeper, 2017-12-20:
Yeah my comment pertained only to your second message, really (that starts with "Apparently more issues."). There is no artefact with your second message that's caused by the mirroring as far as I can see.

We can figure out what objects are mirrored in a project file because the 3MF file stores the original data plus a transformation matrix. Or you could tell us here. However as you said, you are not able to provide these models so you shouldn't provide a project file either then, since that contains the models.

How your walls get broken up after mirroring is a mystery to me. But I do know that the engine can't make up new walls out of thin air. There must be a surface in the middle of the object there.

closed 2018-03-02 by ChrisTerBeke