Ultimaker/Cura #2918: Extruder gets cooled down before end of print

MB3Y, 2017-12-05:
Hi,
I have a diamond hotend, taking 3 filaments in.
On the last layer Cura generates "M104 T2 S0" before the end of the print, i.e. the print is finalising last skin with T0.
On diamond hotend setup though M104 controls the temp for THE extruder, so heating is shut down completely :(
I can surely upload project and gcode if you require it and tell me where to put it.
Cherio,
Mikael

nallath, 2017-12-15:
We don't have a diamond hotend, so this is almost impossible for us to test. We can't provide any support for this unfortunately.

ChrisTerBeke, 2017-12-15:
I think the firmware should actually handle this, Cura is not aware that all hot-ends are on the same extruder (T). The firmware should prevent using T1 or T2 to disable the heating.

Ghostkeeper, 2018-01-02:
You could also (as a workaround) add a Search-and-Replace post-processing script that replaces `M104 S\d+ T` with `;<removed M104>`

MB3Y, 2018-01-02:
I'd love that.. but don't know how to couple a post processor directly into Cura? Is there a user guide as of how to make and couple such yourself?

Ghostkeeper, 2018-01-02:
It's under Extensions -> Post-Processing -> Modify G-Code.

MB3Y, 2018-01-02:
@Ghostkeeper : that was the part I already figured out :) - my questions are about where to put a script, having an example script, how to couple it such that it appears in this list, etc. I assume these scripts are in Python? Are there other alternatives?

MB3Y, 2018-01-14:
Ha... figured @Ghostkeeper 's suggestion out - I thought I had to make myself a plugin to do such thing... Which I still would like, such that I can group and process more Geeetech Rostock 301 "misbehaviors" and "shortcuts".

Goto Extensions -> Post-Procession -> Modify G-Code and select "Search And Replace" as modification.
Next one can do the following for correct replacement:
* Search: "M104 T. S0"
* Replace ";Removed M104 T. S0"
* RegExp: "Yes"
[image]

Another pretty annoying one on the Diamons hotend is all the M109's during the print. If for any reason your temp drops, or your hotend can't fully keep up, then you'll get pauses and oozing on the location of that pause.
The following replacement can help - without interferring too much in the code
* Search: "M109"
* Replace "M104"
* RegExp: "No"
[image]

As this also removes your inital heating up of the nozzle, one should update the start script to include a ";INIT=WAIT FOR S{material_print_temperature_layer_0}" and next a third replacement
[image]
So the following replacement makes such that you keep the necessary command
* Search: ";INIT=WAIT FOR "
* Replace "M109"
* RegExp: "No"
[image]
I didn't want to remove the original M109 S{material_print_temperature_layer_0} as this could have unwanted side-effects as I got pointed out lately that Cura does check if the start script contains the necessary temp preparation commands - else inserts them him/her-self after your own start script.

...Now I hope Cura will remember my favorit replacement throughtout sessions, reboots etc...

MB3Y, 2018-03-07:
@karabas : I don't get your comment? Turning off temps in end_gcode is possible afaik.

karabas, 2018-03-15:
You cannot control turning off prelast extruder. Cura does NOT allow it.

Ghostkeeper, 2018-05-15:
Our project manager took this off of our planning. We'll likely not get around to implementing this soon. For now I suggest to keep all nozzle temperatures the same at your desired printing temperature.

Ghostkeeper, 2019-02-13:
There's two major reasons why the main developers from Ultimaker aren't spending time on this:
* We don't have a printer with this type of nozzle, which you'd need to get an idea on how to make this work elegantly.
* Ultimaker wants us to prioritise issues that have some value for Ultimaker's printers, and this doesn't.

Of course we'll gladly help other people get this in via a pull request, but otherwise I don't see it happening. The issue is still open here on Github, but unless there is no mandate from Ultimaker I won't be able to spend any time on it.

rich2art, 2020-01-20:
Does anyone fixed this Problem?
I tried to change the gcode. Maybe I did something wrong or it's just senseless to make the "search and replace move" with the 104?

smartavionics, 2020-01-20:
I added a setting to the machine settings dialog that let's you specify that the printer's extruders share a heater. I don't think that setting is in the UM releases (yet) but you can install one of my releases to try it out. You can find them at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there.

[image]

mahtDFR, 2020-01-20:
Thanks for the report, this is fixed as of 4.5 so I will close this issue. The fix is the same as smartavionics, using a shared heater checkbox

closed 2020-01-20 by mahtDFR