Ultimaker/Cura #2919: Multi extrude gcode doesn't explicitely select start extruder

MB3Y, 2017-12-05:
Gcode generated for 3 extruders (maybe also 2) doesn't add specifically "T0" as start extruder.
My printer remembers the last used extruder and start print with it.
So my print starts with T0, T1, or T2 in absence of an explicit extruder select cmd.

fieldOfView, 2017-12-05:
Duplicate of #1819 

This will be fixed in Cura 3.2 (the fix in https://github.com/Ultimaker/CuraEngine/pull/640 was not merged in time for Cura 3.1)

closed 2017-12-05 by ChrisTerBeke

MB3Y, 2017-12-05:
Nice!
Unfortunately the workaround indicated in #1819 is not possible for me, since the Geeetech Rostock 301 has mixing features that are to be selected before the print starts. Putting T0 in my start script would overrule this feature.
I'll be patiently awaiting R3.2 ;)

stuartpb, 2018-01-01:
Well, *some* of Ultimaker/CuraEngine#640 must have been merged, because all my multi-extruder G-code starts with `T0` already in 3.1, and I'm not using the RepRap or Griffin flavor.