Ultimaker/Cura #2920: Incorporate support for single nozzle multi extruder

MB3Y, 2017-12-05:
(Ref e.g. Repetier Printer setting "Printer has a mixing extruder (one nozzle for all colors)"

This option should combine nozzle temperature management into one set of commands rather than per extruder.
In Cura 3.0.4 the first layer temp drops when non active extruders are pushed into standby temp (which I have set to the normal printing temp) - and since setting inactive extruder equals setting THE extruder.

This would be less bad, if it also wasn't for the M109 at extruder switch, resulting in stationairy (dripping) nozzle on first layer when switching extruder (selecting first layer print temp again after standby temp was selected and used for printing)

MB3Y, 2017-12-05:
A possible solution to this problem could be to do the temperature settings for the active extruder as LAST action on extruder switches rather than FIRST action in Cura 3.0.4?

ChrisTerBeke, 2017-12-06:
I know one printer that has 2 extruders but 1 nozzle and that firmware actually correctly picks up T0 and T1 commands as specified in Marlin/RepRap. I think that whatever a manufacturer does to their machine, it should always adhere to G-code standards, otherwise we'll be building edge cased in Cura endlessly.

Setting the temperature for active extruder at the end of an extruder switch would break other things, as extruder switch processes often include wiping and purging.

MB3Y, 2017-12-08:
You're telling me that a command like M104 Tx Sy shouldn't change the active tool unless Tx is the active tool? And that the firmware of the printer should take the printer specifics into account?
Damn chinese copycats @Geeetech :( - it's not the only firmware "deviation" that I've encounter or bumped into over the last year :'(

ChrisTerBeke, 2017-12-08:
In case the machine has 2 nozzles it should, but when there's only 1 nozzle, it shouldn't process that as lowering the nozzle temperature as that's obviously wrong for that machine. Chinese are good at cloning hardware, software not so much :P

MB3Y, 2017-12-08:
Thanks for the review anyhow - I shall remember to do some post-processing on my gcode.

closed 2017-12-08 by MB3Y

Ghostkeeper, 2019-02-18:
See also https://github.com/Ultimaker/Cura/issues/4204.