Ultimaker/Cura #2921: Feature Request: Time lapse video capture

fraceman, 2017-12-06:
It would be great if the system could record a time lapse video of the build.  I've had a few failures overnight that it would be great to be able to go back and see when and how it began to fail.  

Not to mention it would be a fun feature to show people how you created their part in 30 seconds.

ChrisTerBeke, 2017-12-06:
I don't think this is something we'll add in core Cura, as you'd have to keep Cura open during the entire print. Someone could write a plugin of course to do this.

Ghostkeeper, 2017-12-19:
Depending on the API of the printer you're using, it could be very easy to make a shell script that fetches the latest camera image from the printer with `wget` and feeds it to `ffmpeg` to encode into a video.

fraceman, 2017-12-19:
Hi Guys,
I appreciate the advisement on doing this via programing plugins.  I am however a Mechanical Engineer and a fossil of a programmer (think DOS, BASIC, FORTRAN circa 1980's).  

I am using the Ultimaker system professionally to produce my designs for concepts and functional tools.  But it's success rate is barely 50%.  It often fails with a peeling of the base layer, but when it occurs I can't determine.  I will watch 10-20 layers lay down well, then failures occur overnight.  I have to send most of my models out to be printed on a Dimension 1200 for about $100 each.  It's not long before we've paid as much for parts going outside as the Ultimaker costed.  Sorry for the rant...   I'd just like this video capture feature to help me solve my problems, and increase the value of our Ultimaker.  I think the feature would be a great sales feature and a tool for fellow pro level users.

Thanks,
Kevin

Ghostkeeper, 2018-01-05:
Try increasing the build plate temperature a bit or other tweaks to improve bed adhesion? Concentric initial layer bottom pattern?

To create a time lapse, try running this command:

    ffmpeg -i http://10.183.1.115:8080/?action=stream -vf select='not(mod(n\,40))',setpts=N/FRAME_RATE/TB timelapse.avi

But replace `10.183.1.115` with the IP address of your printer (visible on the printer under System -> Network) and replace the `40` with the speed increase you want. You may use other ffmpeg arguments to improve the encoding quality because the defaults are crappy. Ideally you'd do a 2-pass.

closed 2018-02-28 by ianpaschal