Ultimaker/Cura #2922: flowrate in cura 3.1

xennonl, 2017-12-06:
hi, after upgrading to cura 3.1 the flowrate is way to low
its set at 100% in the slicer but i need to set it at 260% to have normal flow

in 3.04 it was fine

Hellsing, 2017-12-06:
I can confirm this issue, though 200% flowrate set directly in Marlin seems to be fine aswell (compared to 3.0.4 where I only needed 100% in Marlin)

ChrisTerBeke, 2017-12-06:
Did you use the 3.1 beta before using 3.1 stable?

Hellsing, 2017-12-06:
I did not, fresh install of 3.1 after it popped up in my 3.0.4 today.

Hellsing, 2017-12-06:
Alright, after looking at the screenshot, it seems it has changed my PLA diameter from 1.7 to 2.85mm without me touching it.

Maybe this issue should be addressed.

ChrisTerBeke, 2017-12-06:
Good to hear. Still leaves the question why the diameter suddenly changed though...

Ghostkeeper, 2017-12-19:
Maybe it selected a different material profile which happened to have the same name?

closed 2018-11-26 by Appesteijn