Ultimaker/Cura #2924: Crash error

Piman314159, 2017-12-07:
[image]
So I tried switching printer profiles and all of a sudden around 25 cura windows pop up resulting in it crashing. After trying multiple times I get the same results.

LipuFei, 2017-12-08:
Hi @Piman314159, could you get the log for us? Thanks.

fieldOfView, 2017-12-08:
The 25 cura windows are (empty) Crash Handler windows. I do have the same problem that the crash handler does not succesfully open its window (the crashhandler rework), but always just blamed my from-source-setup.

Ghostkeeper, 2017-12-20:
Switching printers in general doesn't cause this. Which printer specifically did you switch from/to? Maybe you could provide a copy of your configuration folder to allow us to test this.

Piman314159, 2017-12-20:
It happens when I switch from a flashforge creator pro profile to a custom printer. Where would I find the 'log'? 
[Cura.cfg.zip](https://github.com/Ultimaker/Cura/files/1576099/Cura.cfg.zip)
Is this what you mean by config folder?

Piman314159, 2017-12-20:
[Crash log.zip](https://github.com/Ultimaker/Cura/files/1576134/Crash.log.zip)
Alright, I managed to recreate and get the crash log.

LipuFei, 2017-12-20:
@Piman314159 The config folder is `C:\Users\<your-username>\AppData\Roaming\cura\3.1` and the log file `cura.log` is in there too.

Piman314159, 2017-12-22:
[cura.log](https://github.com/Ultimaker/Cura/files/1581343/cura.log)

Ghostkeeper, 2018-01-05:
Some of the relevant parts of the log (LOTS of errors in there):

    2017-12-20 10:38:45,039 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
    2017-12-20 10:38:45,041 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
    2017-12-20 10:38:45,043 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 770, in event
    2017-12-20 10:38:45,044 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 297, in event
    2017-12-20 10:38:45,046 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Event.py", line 189, in call
    2017-12-20 10:38:45,048 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
    2017-12-20 10:38:45,050 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\Settings\SettingInheritanceManager.py", line 99, in _onPropertyChanged
    2017-12-20 10:38:45,052 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\Settings\SettingInheritanceManager.py", line 155, in _settingIsOverwritingInheritance
    2017-12-20 10:38:45,053 - CRITICAL - cura.CrashHandler.__init__ [65]: AttributeError: 'NoneType' object has no attribute 'getProperty'
    2017-12-20 10:38:45,242 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
    2017-12-20 10:38:45,245 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
    2017-12-20 10:38:45,249 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 551, in _onSettingChangeTimerFinished
    2017-12-20 10:38:45,252 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 597, in _updateDisallowedAreas
    2017-12-20 10:38:45,253 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 956, in _getEdgeDisallowedSize
    2017-12-20 10:38:45,255 - CRITICAL - cura.CrashHandler.__init__ [65]: ValueError: max() arg is an empty sequence

I think it's best to delete the custom printer entirely and recreate it. That would at least solve the `max() arg` issue.

closed 2018-02-26 by ianpaschal