Ultimaker/Cura #2927: Front View not working

BagelOrb, 2017-12-07:
Windows 10, GTX Titan, etc.

Front view button doesn't set view to front, but to a different position.

Front view button: 
[image]

Produces:
[image]

Rather than:
[image]

Also:
the angled view button produces unexpected results, but I don't know whether this is intended.

I mean this button: 
[image]

Produces:
[image]

Rather than:
[image]

nallath, 2017-12-15:
@alekseisasin; Could you have a look at this?

alekseisasin, 2017-12-15:
We already have a ticket for this CURA-4642, it will be solved before next realize.

ChrisTerBeke, 2018-01-03:
Fixed in https://github.com/Ultimaker/Cura/commit/5a9e3a8efd0a9e1a69fbcd0c8e4873906bcbcca2. The "3d" view will remain the same, even though it doesn't 100% reflect the icon, but that's just an icon representation.

closed 2018-01-03 by ChrisTerBeke