Ultimaker/Cura #2929: Zonestar p802QR2 extruder 2 not heat

salpanda, 2017-12-07:
On cura 3.1 my zonestar P802QR2 extruder 2 does not heat.
Only extruder 1 heats ok.
Please help
Thank you

ChrisTerBeke, 2017-12-07:
Does it work properly with other slicers? We don't really give support on 3rd part machines unless it's a Cura bug of course.

salpanda, 2017-12-08:
yes with repetier host and slic3r prusa edition is working ok

ChrisTerBeke, 2017-12-08:
Can you send us a project file in which that printer is selected? Then we can debug if it's configuration or an actual bug.

Ghostkeeper, 2017-12-30:
When I slice your project file I get the following g-code at the start:

    T0
    M190 S70
    M104 S230
    M104 T1 S230
    M109 S230
    M109 T1 S230

So it seems that the second extruder (T1) should heat up to 230 degrees.

It seems that your firmware is based on Repetier, so maybe you have to adjust the g-code flavour: https://github.com/repetier/Repetier-Firmware/tree/master/boards/Zonestar%20P802M

That start g-code also contains some variables that are not known to Cura, such as:

    G1 Z15 F{Z_TRAVEL_SPEED}

The setting that comes closest in Cura is `max_feedrate_z_override` a.k.a. Maximum Z Speed. Maybe you wanted to use that? Or does your printer's firmware understand that variable?

closed 2018-03-02 by ChrisTerBeke