Ultimaker/Cura #2930: Cura 3.1 crash on initialization

draput, 2017-12-07:
Hi,

I want to report a crash at initialization at the "Set-up scene..." phase. Windows7 64 bit. Cura 3.0.4. is working without problems.

Best regards,
Draghi

ChrisTerBeke, 2017-12-07:
Did you by any change use the 3.1 beta before this?

igorepi, 2017-12-09:
I have exactly same problem with Cura 3.1 and W7 x64 : 

2017-12-09 13:38:14,596 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Connected
2017-12-09 13:38:14,597 - DEBUG - UM.Backend.Backend._onSocketStateChanged [168]: Backend connected on port 49674
2017-12-09 13:38:24,449 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-12-09 13:38:28,046 - INFO - FirmwareUpdateChecker.FirmwareUpdateChecker._onContainerAdded [52]: You have a 'Tevo Tarantula' in printer list. Let's check the firmware!
2017-12-09 13:38:28,049 - DEBUG - UM.Settings.ContainerRegistry.load [231]: Loading data into container registry took 13.454769611358643 seconds
2017-12-09 13:38:28,581 - DEBUG - UM.Controller.setActiveView [86]: Setting active view to SolidView
2017-12-09 13:38:28,789 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-12-09 13:38:28,791 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-12-09 13:38:28,792 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2017-12-09 13:38:28,794 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2017-12-09 13:38:28,795 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2017-12-09 13:38:28,796 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2017-12-09 13:38:28,798 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "__startup__.py", line 12, in <module>
2017-12-09 13:38:28,799 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2017-12-09 13:38:28,800 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2017-12-09 13:38:28,802 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2017-12-09 13:38:28,803 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2017-12-09 13:38:28,804 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "Console.py", line 24, in <module>
2017-12-09 13:38:28,806 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\bin\cura_app.py", line 83, in <module>
2017-12-09 13:38:28,807 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 681, in run
2017-12-09 13:38:28,809 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 107, in __init__
2017-12-09 13:38:28,810 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 733, in getMachineManager
2017-12-09 13:38:28,811 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 1234, in createMachineManager
2017-12-09 13:38:28,813 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 109, in __init__
2017-12-09 13:38:28,814 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 357, in setActiveMachine
2017-12-09 13:38:28,815 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Application.py", line 156, in setGlobalContainerStack
2017-12-09 13:38:28,817 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
2017-12-09 13:38:28,818 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
2017-12-09 13:38:28,820 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 500, in _onStackChanged
2017-12-09 13:38:28,821 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 597, in _updateDisallowedAreas
2017-12-09 13:38:28,822 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 956, in _getEdgeDisallowedSize
2017-12-09 13:38:28,824 - CRITICAL - cura.CrashHandler.__init__ [65]: ValueError: max() arg is an empty sequence

padizar, 2017-12-09:
Сonfirm this problem

ChrisTerBeke, 2017-12-10:
Again: did any of you use the 3.1 beta before using the 3.1 stable? If so, remove your 3.1 data folder and try 3.1 stable again. If it then still gives error's we'd like the complete log files for reproduction.

igorepi, 2017-12-10:
No, I don't use 3.1 beta. 
[cura.log](https://github.com/Ultimaker/Cura/files/1545966/cura.log)

draput, 2017-12-10:
No, I did NOT used the 3.1 beta. I have used 3.0.4.

ChrisTerBeke, 2017-12-10:
@LipuFei related to that fix you did on Friday?

ChrisTerBeke, 2017-12-10:
Related to https://github.com/Ultimaker/Cura/issues/2798?

ChrisTerBeke, 2017-12-10:
I'm closing this issue in favor of the already existing https://github.com/Ultimaker/Cura/issues/2798. Please keep an eye on that one to track the status of fixing this.

closed 2017-12-10 by ChrisTerBeke