Ultimaker/Cura #2931: Enabled support with initial layer height of 0.1 and support confuses build plate adhesion generation

gregor4005, 2017-12-07:
I tried to slice the file Filament-Spool-Mount-Universal_-_Base.stl from https://www.thingiverse.com/thing:2179768 and recognized, that with an _Initial Layer Height_ of 0.1 and with enabled _Support_ confuses the slicing engine and produces a _Build Plate Adhesion_ only around the Support-Material.
[image]

ChrisTerBeke, 2017-12-07:
Can you remove all your 3.1 user data and try the 3.1 stable that came out 2 days ago?

Ghostkeeper, 2017-12-20:
This looks suspiciously like https://github.com/Ultimaker/Cura/issues/1229, but it's not the same since this seems to have something to do with support (which was disabled in that issue).

ChrisTerBeke, 2017-12-20:
We had this in the office last week. The problem then was the model (it was broken on the bottom). After repairing with netfabb the skirt was correct.

Ghostkeeper, 2017-12-20:
So in the engine it's possibly using a different polygon for generating the skin than the one it uses for generating the skirt.

ChrisTerBeke, 2017-12-20:
Probably, see CURA-4718 for details.

Here an image from before and after fixing:

[image]

closed 2018-03-02 by ChrisTerBeke