Ultimaker/Cura #2932: Problem when change the size layer Cura 3.1.0

ajmarmar, 2017-12-07:
Hi,

When I try to change the layer size Cura, Cura doesn't recalculate the process and can't to save.

[CuraEngine_2017-12-07-200410_MacBook-Pro-de-Andres copia.crash.txt](https://github.com/Ultimaker/Cura/files/1540358/CuraEngine_2017-12-07-200410_MacBook-Pro-de-Andres.copia.crash.txt)

LipuFei, 2017-12-08:
Hi @ajmarmar , if possible, could you share the model(s) you were trying to slice and settings you were using? Thanks.

ajmarmar, 2017-12-08:
the model that I try to print is https://www.thingiverse.com/thing:1190968. When change the layer from 0.2 to 0.3 don't slicer.
I attached a video for explain me.
[CE_Video_1512674358030.mp4.zip](https://github.com/Ultimaker/Cura/files/1543143/CE_Video_1512674358030.mp4.zip)

LipuFei, 2017-12-11:
@ajmarmar Could you save it as a project file and upload it? You can do that via the menu "File" -> "Save as project".

ajmarmar, 2017-12-11:
Hi,
I attached the project.
[A_ArduinoBreadBoard_issue.3mf.zip](https://github.com/Ultimaker/Cura/files/1547930/A_ArduinoBreadBoard_issue.3mf.zip)

LipuFei, 2017-12-11:
@ajmarmar Looks like there's a problem with the "Ironing" feature. If you disable that for now, your model will be sliced. I will work on a fix for that.

It's in category "Shell":
[image]

LipuFei, 2017-12-15:
Hi @ajmarmar , turns out a crash can occur if "Ironing" and "Remove Empty First Layers" are both enabled. I made a fix for it: https://github.com/Ultimaker/CuraEngine/commit/e79893968d5c6497b7d1c0e1015722b07f38b3b5

closed 2017-12-15 by LipuFei