Ultimaker/Cura #2933: Cura 3.1.0 does not respect nozzle size

thattolleyguy, 2017-12-07:
When I try and change my nozzle size to .2 mm on my printrbot, the line width is not recalculated to .2 mm, it remains .4 mm. If I try and set it to .2 mm the field is highlighted with a warning. I've tried to get a print to work correctly but I haven't been able to.

ChrisTerBeke, 2017-12-08:
Hi. Did you use the 3.1 beta before the 3.1 stable?

jbloggz, 2017-12-08:
I have the same issue. Tested on both Linux (Ubuntu 16.10) and windows. 
Steps I did:
   1. Install Cura 3.1 (Release date 12/5/2017)
   2. Create and activate new printer (Other -> Tevo Tarantula)
   3. Open "machine settings" and set Nozzle size to 0.3 (was 0.4 previously)
   4. Close window and see that "line width" is still set at 0.4

On my previous version (2.6.2), it would update the line width to 0.3, but not on 3.1

jbloggz, 2017-12-08:
I just tried v3.0.4 (Release date: 10/17/17) and it works fine. With the same steps as above, the line width changes to match the nozzle size.

fieldOfView, 2017-12-08:
Confirmed.

This is a regression caused by the recent rework of single-extrusion printers. The nozzle size is being set in the global stack, whereas it should be set in the (newly introduced) extruder stack.

LipuFei, 2017-12-08:
@fieldOfView @jbloggz Thanks for the info! I've made a fix: b4e7216f5

fieldOfView, 2017-12-08:
That is a fairly ugly fix, but it works. It gives the user a number of settings that are basically guaranteed to do nothing on a single extrusion printer.

Here is a work in progress on another fix here: https://github.com/Ultimaker/Cura/compare/master...fieldOfView:fix_machinesettings_nozzle_size

LipuFei, 2017-12-08:
I will wait for that fix. @fieldOfView thanks.

lDrlEli, 2018-01-20:
So when will this be fixed?

fieldOfView, 2018-01-20:
In 3.2

fieldOfView, 2018-01-20:
3.2 should go in beta this week

NexiTech, 2018-02-24:
I have the same problem. I set up Anycubic Kossel Linear plus under profile Deltabot 3D printer in CURA 3.21 and still I can't change nozzle size. Its always 0.5mm and when I change to 0.4mm and save it I check again and its still 0.5mm! Its very frustrating! Please fix this problem :(

sellinie, 2018-02-27:
Same bug here. Nozzle size always reset to 0.5 after closing and opening cura. CURA 3.2.1 Windows 10

Ghostkeeper, 2018-03-21:
It works currently in 3.3 for slicing, but the actual setting won't update until you close the category and reopen the category again. Reproduce steps:

* Add Printrbot Simple.
* Load any model.
* In the settings search bar, type "Line Width".
* In the Machine Settings dialogue, go to the extruder and change the nozzle size from 0.4mm to 0.2mm.
* Close the Machine Settings dialogue.

Now the model reslices and you can see in Layer View that the line width is actually 0.2mm. But the Line Width settings still say 0.4mm. Only when you close the category "Quality" and reopen the category again does it show the proper setting value 0.2mm.

fieldOfView, 2018-03-21:
Does this depend on the Line Width field having the active focus?

Ghostkeeper, 2018-03-21:
No, all of the line width settings (also the inherited ones) don't update.

SirRodney42, 2018-05-14:
Keeps happening with 3.3.1. Is there a workaround until a fix is released?

Ghostkeeper, 2018-06-13:
I can't reproduce this with the reproduce steps that I posted on the 21st of March. The line width is 0.2mm in the print (layer view) as well as displaying 0.2mm in the interface even before closing and re-opening the category.

diegopradogesto, 2018-06-28:
Similar scenario is in #3773.

Ghostkeeper, 2018-06-29:
I think this is fixed in 3.4.0. It's not reproducible any more.

diegopradogesto, 2018-07-26:
Which printer definition are you using? Can you send us a project file (File -> Save Project)

chsc, 2018-10-02:
I use 3.4.1 with a _Delta Bot_ configuration and the problem still exist. Even deleting all configuration files in ~/.config and ~/.local didn't work.

fieldOfView, 2018-10-02:
If it still reproduces, an easy workaround is to set the nozzle size to 0.40001 (or something similar that is no 0.4).

Theoristos, 2018-12-02:
Cura 3.4.1, German Reprap Neo profile. 
Completely the same issue, 0.5 mm instead of 0.4 on every restart.

Theoristos, 2018-12-11:
Err... what is correct? Of course, there is some default value, but the user must be able to change it, and _save_ it. The last is a problem.
One can change nozzle or put spare extruder, but can't save the correct value in soft...

diegopradogesto, 2019-09-10:
I have tested it in the latest stable version (4.2) and in the current beta (4.3 beta) and it seems to work for both German Reprap Neo and Printrbot. So it's apparently fixed now.

closed 2019-09-10 by diegopradogesto