Ultimaker/Cura #2934: Cura 3.1 can't import Cura 3.0 printers and profiles (MacOS High Sierra)

boksbox, 2017-12-08:
I just got the new Cura 3.1.

It has not carried over the old 3.0 printers and profiles. 
Import profiles can't find previous profiles as well

ChrisTerBeke, 2017-12-08:
High. Did you use the 3.1 beta before this?

boksbox, 2017-12-08:
No.  I just downloaded it when I got the email from Cura that the new 3.1 is out.

ChrisTerBeke, 2017-12-08:
@LipuFei

LipuFei, 2017-12-08:
@boksbox Which printer are you using? and could you send the profiles you want to import to us?

boksbox, 2017-12-08:
@LipuFei I'm using a RepRap Kossel Mini with repetier firmware.  The 3.1 managed to get the previous Materials profile but not the old print profiles and printer settings.  
Where can I find the old profiles?

LipuFei, 2017-12-08:
If you are on Windows, you can find them in `C:\Users\<your-username>\AppData\Roaming\cura`.

LipuFei, 2017-12-08:
Ah sorry... Forgot you are on Mac. I got lost between issues...
You can find it in `/Users/<your-username>/Library/Application Support/cura`

boksbox, 2017-12-08:
Ok, I'm already in /Users/<your-username>/Library/Application Support/cura folder.  Where do I find it? Can't find any file with .curaprofile extension.

[image]

LipuFei, 2017-12-08:
They are not saved as `.curaprofile`. Could you upload your `3.0` and `3.1` folders? We can check what failed during the upgrade.

boksbox, 2017-12-08:
I tried uploading the .zipped folders, but they exceeded the 10mb limit here (72mb for the 2 folders)

PianoMan2112, 2017-12-08:
I have the same issue - if I import one of my saved exported custom material profiles, it says it imported, but does not show. (my 3.0 & 3.1 folders compress to 29 MB).

LipuFei, 2017-12-11:
@boksbox @PianoMan2112 This can be a bit confusing, but the material list only shows the materials that are compatible with the machine you have activated. Could this be the case?

boksbox, 2017-12-11:
As I mentioned, my previous machine's settings were not carried over but previously, I only had 1 material and this was carried over from 3.0 to 3.1.  I don't know about the other preset materials because I don't use them.

LipuFei, 2017-12-11:
@boksbox It must be the oversized log file. Could you try removing `cura.log` and `cache` and upload again? Thanks.

boksbox, 2017-12-11:
If I remove just the cache and cura.log they are still over 12mb when zipped. So...

Inside the 3.0 folder there are 2 folders named 2.0 and 15.04.6.  I also removed them together with the cache and cura.log

And in the 3.1 folder I did the same.  I left the 3.0 folder within but took out the cache and cura log inside this 3.0 folder as well.

Is that ok?

[3.0.zip](https://github.com/Ultimaker/Cura/files/1547794/3.0.zip)
[3.1.zip](https://github.com/Ultimaker/Cura/files/1547795/3.1.zip)

LipuFei, 2017-12-11:
@boksbox That's good! Thanks!

LipuFei, 2017-12-11:
@boksbox I made a fix for this c6cdc1d1dbb649b73e66d93fe8584cf57dfcb2cf. The problem is that, if you have a profile with special characters such as "%" in its name, the version upgrade will fail.

To fix this for you, some manual steps are required. You can follow the instructions below:
1. Download this ZIP file [fix_3.1_boksbox.zip](https://github.com/Ultimaker/Cura/files/1547989/fix_3.1_boksbox.zip)
2. You can find two files in this ZIP, those are the upgraded quality files of yours.
3. Copy and paste those two files into your `~/Library/Application Support/cura/3.1/qualities`, one of your "old" files will be overwritten and the other is new.
4. Try start Cura 3.1 again.

Hope this works for you.

LipuFei, 2017-12-11:
Hi @PianoMan2112 Could you provide more information on your issue?
1. Which machine are you using?
2. If possible, could you provide the profile you were trying to import?

Thanks

boksbox, 2017-12-11:
@LipuFei Thanks a lot. I did as you instructed.  
I managed to get the Slow 0% to appear on the Custom profiles but the rest from the previous 3.0 didn't make it.
[image]

DanalEstes, 2017-12-11:
I converted from 3.0.4 to 3.1.0, via the "Ultimaker Cura-3.1.0-win64.exe" installer downloaded 12/10/2017.

All printers came forward.  All materials came forward. All customized print profiles did not. 

Several "Custom" printers (3rd party).   Windows 10 Home, V1703, Build 15063.726.

[cura.zip](https://github.com/Ultimaker/Cura/files/1548266/cura.zip)

LipuFei, 2017-12-12:
Hi @DanalEstes, I have manually recovered the custom qualities for you. You can download this ZIP [fix-DanalEstes-cura3.1-qualities.zip](https://github.com/Ultimaker/Cura/files/1551309/fix-DanalEstes-cura3.1-qualities.zip) and copy it to your `AppData/Roaming/cura/3.1/quality/` directory. Could you try that? Thanks.

LipuFei, 2017-12-12:
Hi @boksbox , somehow I can see your custom profiles after the Cura 3.1 upgrade. Here is a package of all the upgraded quality files [fix_boksbox_cura3.1_quality.zip](https://github.com/Ultimaker/Cura/files/1551373/fix_boksbox_cura3.1_quality.zip). You can try it out by copying and pasting to your `~/Library/Application Support/cura/3.1/qualities` directory.

DanalEstes, 2017-12-12:
Fantastic!  Thank you!

I was not anticipating such great service; I had mostly provided the zip of
the directory so that you folks could avoid this in the future.

Thank you once again,

Danal

boksbox, 2017-12-13:
@LipuFei, That did it!  Thank you very much!

DanalEstes, 2017-12-13:
And, Lipu Fei, that did the trick.  Worked perfectly.

I know I said "thank you" several times above... but... Thank You!

LipuFei, 2017-12-13:
@boksbox @DanalEstes 😄

LipuFei, 2017-12-13:
@PianoMan2112 I will close this issue for now. If you still have that issue, feel free to reopen it.

closed 2017-12-13 by LipuFei

mzhu1113, 2017-12-13:
Hello LipuFei,

I still have this issue and I've tried taking your changes and compiling it and shoehorning it into my App directory (Windows 10) and only 2 of my profiles make it into 3.1. All printers and materials upgrade properly.

Any help would be appreciated as I love the sound of the new features but I can't use them :(

Thanks!

I've attached both 3.0 which works fine and 3.1 which only migrated two profiles.
[3.0.zip](https://github.com/Ultimaker/Cura/files/1556426/3.0.zip)
[3.1.zip](https://github.com/Ultimaker/Cura/files/1556430/3.1.zip)

LipuFei, 2017-12-15:
Hi @mzhu1113 Sorry that I kept you waiting. Your recovered profiles are in 
[fix_3.1_quality_mzhu1113.zip](https://github.com/Ultimaker/Cura/files/1561918/fix_3.1_quality_mzhu1113.zip). Unpack the ZIP and copy-and-paste them to your `C:\Users\<your-username>\AppData\Roaming\cura\3.1\quality` directory. Your profiles should be back. :)

LipuFei, 2017-12-15:
@mzhu1113 Oh, and also remove those files named as "msu%2B-%2Bduplicator", which have those `%xx` escape characters.

mzhu1113, 2017-12-16:
Hey @LipuFei,

No worries. :) I appreciate the help. Not really strong in python programming but it was interesting when trying.

Much appreciated, the profiles are all back. Looking forward to the new features!

Thanks!

PianoMan2112, 2018-01-02:
@LipuFei I'm so sorry; I didn't see any replies and just happened to see these now.  My printer is a Monoprice Mini (relabeled Malyan M200), which isn't in the printer list, so I pick Custom.  The profiles I were trying to import should show in my old version as brand Hatchbox, a sample is:
      <brand>Hatchbox</brand>
      <material>PLA</material>
      <color>White</color>
      <label>White PLA</label>
Note: I had previously exported this as a material file; importing that file appears to do nothing in 3.1.

laegion420, 2018-03-14:
I'm having the same problem now with importing profiles from 3.0.4 into 3.2! Only one made it successfully into the new version.

How did you fix those files? Can someone fix mine for me or tell me how it's done?

DanalEstes, 2018-03-14:
Just for the record:  3.1 to 3.2 dropped all my profiles.  I chose to
rebuild them.

laegion420, 2018-03-14:
The nice part is that you can go to the profile storage location for the previous version and open the profiles in notepad and read all the settings out of them.

ianpaschal, 2018-03-15:
@laegion420 do you get an error popup when trying to import the profile?

laegion420, 2018-03-15:
Yes, I do.

ianpaschal, 2018-03-15:
@laegion420 ...and what does it say?