Ultimaker/Cura #2935: Scroll bar not accesible

Wouw, 2017-12-08:
When you change setting for example from the support, and you close the tab.
It happens that when your are to far scrolld down that you cant scroll back up to matarials becaus the scroll bar has vanishd.

Ghostkeeper, 2017-12-20:
It works for me. See this small video: [scrollbar.zip](https://github.com/Ultimaker/Cura/files/1575047/scrollbar.zip)

It automatically scrolls back up.

Wouw, 2017-12-20:
[image]

when i have multiple tabs open and close the largest one, de top most tabs are almost always gone.
i have to reopen a tab and scroll back up or close the window to reset everything and make the scroll bare visible again.

alekseisasin, 2017-12-20:
This issue a bit difficult to reproduce, but I found the way how to do it. Thank you @Wouw 
Devs: CURA-4732

Ghostkeeper, 2017-12-29:
Using alekseisasin's reproduce steps I could reproduce it too. Or basically a simpler version:
* Expand a category.
* Expand a different category.
* Collapse the first category.
* Position the scroll bar such that it's not entirely at the top (so scroll down slightly).
* Collapse the second category.

I made a fix that will be included in version 3.2.

commit 8b504c8a (2017-12-29): Fix scroll position after collapsing category not in focus

When collapsing a category, the height of the category is briefly set to 0 and then animated there properly. If the setting list becomes so small then that the scroll bar disappears, the view should automatically get scrolled up. But because the focus changed at the same time it would move the scrollbar such that the category header was in view. This clashed, causing the scroll position to end up not completely at the top even though the scroll bar did disappear. Setting the focus after collapsing the scroll bar prevents this.

Contri [... cut]
closed 2017-12-29 by Ghostkeeper