Ultimaker/Cura #2936: Cura Crash on Create New Profile

ChrisRiddell, 2017-12-08:
Cura 3.1
Windows 10
Ultimaker 2+

Cura crashed on me when I tried to "create profile from current settings/overrides"

[cura.log](https://github.com/Ultimaker/Cura/files/1542370/cura.log)

LipuFei, 2017-12-08:
Hi @ChrisRiddell, if you have used Cura 3.1 BETA, please make sure your cura 3.1 beta directory is removed. If that's not the case, could you send your cura directory to us? I cannot find anything useful in the log...

ChrisRiddell, 2017-12-08:
Fresh install of Cura 3.1 all appdata folders where removed from beta before I got this.

Do you want the one from C:/ programs or the folder from appdata?

LipuFei, 2017-12-08:
From `AppData\Roaming\cura`. Thanks!

ChrisRiddell, 2017-12-08:
[cura.zip](https://github.com/Ultimaker/Cura/files/1542698/cura.zip)

There you are hope it helps

LipuFei, 2017-12-08:
Hi @ChrisRiddell , sorry that I cannot reproduce this problem with and without your cura settings. Do you still get this crash?

ChrisRiddell, 2017-12-08:
Only had the crash twice it seems to be super random if it happens or not when making the new profile.

ChrisRiddell, 2017-12-09:
Can't reproduce this no more not sure what was happening but seems to of fixed it's self.

closed 2017-12-09 by ChrisRiddell