Ultimaker/Cura #2937: When changing settings in Cura 3.1 after slicing are restored

rmcbc, 2017-12-08:
When I'm using customized settings for example changing infill from 25% to 40%, after slicing starts the number is restored to 25% again and starts slicing again.
It happens in several places in other settings the one mentioned is only an example.

ChrisTerBeke, 2017-12-08:
Did you use the 3.1 beta before this?

rmcbc, 2017-12-08:
Yes.

ChrisTerBeke, 2017-12-08:
Can you try removing the 3.1 user data folders and then start Cura 3.1? It should then start picking up from the 3.0 config correctly.

rmcbc, 2017-12-08:
I delete this folder:
C:\Users\ruica\AppData\Roaming\cura\3.1
Is this the correct one?
It seems that the problem is fixed.

ChrisTerBeke, 2017-12-08:
That's the one yes. Good to hear that fixes your issue. It was a bug in the beta so anyone that used the beta should remove their user data before using stable. Thanks for testing with me!

closed 2017-12-08 by ChrisTerBeke

rmcbc, 2017-12-08:
Thanks for your support.