Ultimaker/Cura #2938: Pointless travel moves?

ChrisRiddell, 2017-12-08:
Cura 3.1
Windows 10
Ultimaker 2+

I noticed when doing some prints there seems to be some pointless travel moves

https://www.youtube.com/watch?v=HheyxG9LGeY

Here is a project file if that's any help to see what I'm seeing.
[UM2+_3DBenchy.curaproject.zip](https://github.com/Ultimaker/Cura/files/1542692/UM2._3DBenchy.curaproject.zip)

smartavionics, 2017-12-08:
Hi, those moves are because you have combing set to "no skin" and because those regions that are being filled are skin it has to "go the long way round" when moving from one side of the region to the other. If you turn off combing altogether it will travel directly but will always retract. If you set the combing mode to "all" it will also travel directly but could leave a line across the skin surface.

ChrisRiddell, 2017-12-09:
Okay makes sense,

With combing mode to "all" their still seems to be some not very optimized travel moves.

in this video you will see it finishes the outline goes to the other side of the model to finish the layer but then comes all the way back to the front to finish doing 2 little lines then goes to the infill. I think it would make more sense to do them 2 little lines before moving to the back of the model so it does not have to travel all the way back to where it was.

https://www.youtube.com/watch?v=1XJpIs9TalQ&feature=youtu.be

smartavionics, 2017-12-09:
I can't really work out what it's doing from that video clip but here's a general point. Cura does not currently try to finish everything (walls, infill, skin, support, etc.) within a local region before moving to another region. It will do all the walls, all the infill and all the skin as groups. For some models and settings it will travel more than it really needs to due to that ordering. i should think that there would be some difficult issues involved in interleaving the printing of walls/skin/infill so I doubt if this is going to be changed anytime soon (it's certainly not on my to do list). Version 3.2 of Cura should incorporate some changes that improve the order used for printing infill lines and skin lines so that will help a little.

closed 2017-12-09 by ChrisRiddell