Ultimaker/Cura #2941: Not needed support

fabus1989, 2017-12-08:
Hello, 

I've testing cura 3.1.0 and it generates not needed support (screenshot attached).

Please let me know if any further information is needed.

Best regards
Fabian
[image]

ChrisTerBeke, 2017-12-08:
Might be a bug in the engine. Can you upload a project file with this model and the settings so we can reproduce it?

fabus1989, 2017-12-08:
It's not possible here. Maybe you will share your email?

fabus1989, 2017-12-08:
Sorry I forgot about cloud services.

Here is the link to model: [https://1drv.ms/u/s!AnrstGN8arUcj89d4GjluhtoyJzVMA](url)

ChrisRiddell, 2017-12-08:
Go to custom settings and turn off Support Towers, I had the same thing.

fabus1989, 2017-12-08:
Thanks for workaround :)

nallath, 2017-12-15:
Is your horizontal expansion of the support set to non zero?

Ghostkeeper, 2017-12-20:
This is a duplicate of https://github.com/Ultimaker/CuraEngine/issues/544.

closed 2017-12-20 by Ghostkeeper

alekseisasin, 2018-01-25:
Hello @fabus1989, Sorry for long time waiting reply, for now this issue can be solved if you will set setting
¨Support Horizontal Expansion¨ value to 0.