Ultimaker/Cura #2942: Cura 3.1 extruder temp problem

bhamlin52, 2017-12-08:
Cura 3.1 Printing Temperature will change in the gcode. After 4 stringing test slices Cura 3.1 with temp set at 195 or 200 the gcode was at 210. 

[image]

;FLAVOR:Marlin
;TIME:13575
;Filament used: 8.69164m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.1.0
M82 ; absolute extrusion mode
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S210 ; set extruder temp
M140 S50.0 ; set bed temp
M190 S50.0 ; wait for bed temp
M109 S210 ; wait for extruder temp

Liukkjan, 2017-12-08:
I am having this issue as well. Changing printing temperature does not affect saved Gcode. Has the {variables} used in start up code changed?

fiveseven808, 2017-12-09:
I have this issue. So freaking annoying. I can no longer set my first layer temperature. It defaults to 210, I need to edit the gcode manually or change it via Octoprint while it's going.

bhamlin52, 2017-12-09:
I finally uninstalled 3.1 and went back to the prior version. When I did that I got all my profiles back also. 3.1 so far is totally useless.

notnyt, 2017-12-09:
I have this same issue,   Both of these below always output to 210, no matter what I set it to.
````
M104 S{material_print_temperature_layer_0}
M104 S{material_print_temperature}
````

fieldOfView, 2017-12-09:
I would wager a bet that this is a regression caused by the single extrusion rework. Devs: I think the replacement pattern variables are taken from the global stack, but they should be taken from an extruder stack now. Perhaps the easiest way to fix this is to add a resolvement function on the global stack that resolves to taking the value from the first active extruder.

joker3d, 2017-12-09:
i have same problem with a new update

ChrisTerBeke, 2017-12-10:
Did any of you use the beta 3.1 version? If so, remove the 3.1 data folder and then start the 3.1 stable (there were some issues in the beta with project files and version upgrade data storage).

notnyt, 2017-12-10:
I completely uninstalled and removed all of the data folders and the issue persisted. any of the temperature variables in my start script get replaced with 210

bhamlin52, 2017-12-10:
I never installed the 3.1 beta version.

ChrisTerBeke, 2017-12-10:
Okay, clear. I think @fieldOfView's comment is very relevant, I'll mention it to the team tomorrow and create an internal ticket after that.

bhamlin52, 2017-12-10:
Thanks for taking a look at this. You should be able to duplicate. Just slice something and check the gcode.

Ghostkeeper, 2017-12-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/1296, https://github.com/Ultimaker/Cura/issues/2091, https://github.com/Ultimaker/Cura/issues/2546 and https://github.com/Ultimaker/Cura/issues/2812. Only now it also manifests on single extrusion printers because those printers now also have an extruder stack.

closed 2017-12-10 by Ghostkeeper

ChrisTerBeke, 2017-12-15:
@fieldOfView I'm working on fixing this issue now, will probably be similar to your suggestion, thanks!

tlhintoq, 2018-01-14:
So this ticket is marked 'closed' - Any idea of when a new release will be available?  Including fix on single-extruders?  I have all previous versions installed side-by-side so right now I just use 3.04 instead of 3.10. The bug isn't killing me for that reason; but its nice to have all the little updates.

ChrisTerBeke, 2018-01-14:
The beta for 3.2 is due in ~1.5 weeks. Keep in mind that it's a beta, so other things might not be working 100% as intended. The stable release will follow about 2 weeks after that.

kelemenlajos, 2018-02-07:
I think i found something. If you install the app, and don't change the material, only switching to custom setup, the extruder temp works as expected. But after changing Material the "Printing Temperature" value isn't used anymore.

Could someone also test this?
(I reproduced the problem on my home and work computer.)