Ultimaker/Cura #2944: new 3.1 prints items several layers off the raft

conroymp, 2017-12-09:
Several support layers will print, but the first model layer is not over support and off the raft many times the  air gap.  Brim and skirt are fine.  Was in 3.0 as well.  

By layer, prints raft (3 layers), 2 layers of support, then piece (1 layer), then starts printing both on layer 8.

nallath, 2017-12-15:
Could you share projects in which this happens? This is not enough info for us to pick it up.

conroymp, 2017-12-20:
Jamie,

Sure, that attached file, when printed on its side, on a raft caused the problem.

Mike

From: Jaime van Kessel
Sent: Friday, December 15, 2017 7:37 AM
To: Ultimaker/Cura
Cc: Mike Conroy; Author
Subject: Re: [Ultimaker/Cura] new 3.1 prints items several layers off the raft(#2944)

Could you share projects in which this happens? This is not enough info for us to pick it up.

nallath, 2017-12-20:
You only sent a mail, which fails with attachments on github. Could you please upload it via github?

Ghostkeeper, 2018-01-05:
Here's what happens with default settings on an Ultimaker 3 (layer height 0.1mm, raft air gap 0.3mm):

* 4 raft layers.
* 4 layers of only support (seems to be 1 layer too many).
* The rest onwards is model+support.

conroymp, 2018-01-23:
What Ghostkeeper said.   

Seems related to the first layer, elephant foot scaling.  Cannot prove it though.   Take the attached file, place the item on its side, on a raft, and slice.  Look at the results.  There will be a missing layer for a long edge.  I cannot install cura on the machine I have at this time.  

[Fence Latch.zip](https://github.com/Ultimaker/Cura/files/1657225/Fence.Latch.zip)

closed 2018-01-23 by conroymp

RagingRyno, 2018-02-04:
I'm having the same issue atm, did you find a fix for it?

Ghostkeeper, 2018-02-26:
I don't think we fixed this, no.

ianpaschal, 2018-02-26:
No, it was the OP who closed it. @RagingRyno are you still experiencing this? If so, I will reopen this. (Would rather reopen than create a new issue.)