Ultimaker/Cura #2946: Update to Cura 3.0.4 wrongfully changes material diameter

couka, 2017-12-09:
I just updated from Cura 2.x.x to 3.0.4 and noticed serious underextrusion.
I found that the material diameter for my machine (Anet A8) was wrongfully changed from 1.75 to 2.85mm for no reason.

A working quick fix for already exported gcode is setting the extrusion multiplier to 265% (assuming 100% was fine before)

ChrisTerBeke, 2017-12-10:
Does changing the diameter back to 1.75 fix the issue as well? Not sure about the version upgrade itself as we don't test on any 3rd party machines.

ChrisTerBeke, 2017-12-10:
Related to https://github.com/Ultimaker/Cura/issues/2922?

LipuFei, 2017-12-12:
For reference, I have created a JIRA ticket for this: CURA-4708

Ghostkeeper, 2017-12-19:
This is a duplicate of #2922.

closed 2017-12-19 by Ghostkeeper