Ultimaker/Cura #2947: Feature Request: Slice after n seconds of inacivity

thattolleyguy, 2017-12-09:
I like that cura has added the option to disable automatic slicing. It's great for times when I've got a bunch of small prints that I'm trying to add and arrange on the build plate. However, it becomes a pain when I only have one thing on the build plate and constantly have to push "prepare" every time I change a setting. What would be really nice is a setting to slice after n seconds of no changes. This way if I'm making lots of changes, the slicer waits until a stable state and then automatically slices.

Thisismydigitalself, 2017-12-09:
Read this https://github.com/Ultimaker/Cura/issues/2887

Ghostkeeper, 2017-12-13:
Actually currently we slice after 500ms of inactivity:

https://github.com/Ultimaker/Cura/blob/88af41769e4feb690b1561c5ab14a9b675c9a369/plugins/CuraEngineBackend/CuraEngineBackend.py#L138

Otherwise it'll start slicing while you're typing stuff or moving stuff on the build plate. We could fairly easily extend that to 10 seconds if desired.

thattolleyguy, 2017-12-13:
@Ghostkeeper I guess I'm not super familiar with the code. I think making this tunable or increasing it to at least 1-2 seconds makes sense. I know there's a lot of times I'm moving stuff around and it will start slicing when I'm still moving things.

Ghostkeeper, 2017-12-13:
Yes, I'm making a ticket to put a textbox behind the autoslice preference to say "Slice automatically after ... seconds".

fieldOfView, 2017-12-14:
Note that setting that value to 10 seconds would also mean you *have to* wait 10 seconds after each change, without a button to skip those 10 seconds.

Ghostkeeper, 2018-02-21:
This was rejected by our project manager. The preference is very specific and too fine-grained, so it was not seen as good for user experience.

closed 2018-02-21 by Ghostkeeper