Ultimaker/Cura #2948: CuraEngine.exe stops working very frequently

WClampitt1, 2017-12-09:
I am running Cura 3.1. This issue was also happening in the previous version. Whenever I load a model into Cura, I will get a pop-up that CuraEngine.exe is no longer working. They are very simple models that I made in Autodesk Inventor, mostly rectangular in shape.
[image]

thopiekar, 2017-12-09:
Is it possible to share your model files? Would ease to reproduce the issue.

(Oh, and since you are using Inventor: https://github.com/thopiekar/CuraInventorPlugin)

WClampitt1, 2017-12-10:
> CuraInventorPlugin)

Yeah no problem. I had three of each of these.

thopiekar, 2017-12-10:
If you added the model files to your mail, then they have been filtered out by GitHub.
Could you zip them and upload them via browser?

WClampitt1, 2017-12-10:
Here they are.
[Bifold_Flats.zip](https://github.com/Ultimaker/Cura/files/1545988/Bifold_Flats.zip)

thopiekar, 2017-12-10:
Seems to be exported into the wrong unit. Could you tell us the scaling factor of both files?

WClampitt1, 2017-12-10:
Oh yes, sorry I forgot about that.  The longest part of each model should
be 8 inches (203.4mm). So that would be 1000% for Bifold_Flat_1 and 2540%
for Bifold_Flat_2

LipuFei, 2017-12-13:
@WClampitt1 Could you also save and share it as a project file? You can do that via menu "File -> Save project". That way all your models and settings will be saved altogether.

WClampitt1, 2017-12-13:
Yeah, no problem!
[CC_Bifold_Flats.curaproject.zip](https://github.com/Ultimaker/Cura/files/1555977/CC_Bifold_Flats.curaproject.zip)

g3gg0, 2017-12-13:
having a similar problem. 
cura is crashing very often, depending on the settings affecting layer height.
e.g. if configured 0.15mm which worked, swtiching to 0.2mm could cause a crash.

got a model which permanently crashes.

here a .3mf of my setup
https://1drv.ms/u/s!Amx_Fd-zOoXnnbgoFo-NSqyyguqKmA

the last message in stderr.log appears to be:
2017-12-13 20:54:35,740 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] [WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.

with a lot of
[WARNING] Polygon has multiple verts at the same place: (140105, 144487); PolygonProximityLinker fails in such a case!
in front of that message.

LipuFei, 2017-12-15:
@WClampitt1 I could not reproduce the crash unfortunately. I tried moving things around and using different layer heights, but they all worked for me.

Ghostkeeper, 2018-01-02:
The warning of multiple vertices in the same place can be ignored. It's an indication that the wall overlap compensation might be wrong, but it should just continue slicing.

I could also not reproduce the crash, neither with the CC_Bifold_Flats project, nor with g3gg0's Voronoi bowl. It slices fine.

g3gg0, 2018-01-02:
here a screenshot of it crashing with the 3mf file i posted.
is there anything i can upload too, so you can reproduce the crash?

Ghostkeeper, 2018-01-18:
If I go back to the Cura 3.1 release it indeed crashes! But not in 3.2 any more. Maybe we could wait until the beta of 3.2 is released and see if that solves your problem.

ianpaschal, 2018-03-09:
@g3gg0 @WClampitt1 : Still having this problem in 3.2.1?

g3gg0, 2018-03-09:
didn't face any similar issue since 3.2.x

closed 2018-03-09 by ianpaschal